Ultimaker/Cura #3326: SweepAZees (a TweakAtZ based gcode post processing script)

JJJpnt, 2018-02-15:
**Cura 2+**

**All platforms**

Hello, first sorry I don't know if it's appropriate to post that here, feel free to delete if it isn't.

→ I modified Cura's post processing plugin TweakAtZ (gcode post processing script which use is to tweak a parameter at a given layer/height), so it does several parameter tweaks over a given range instead of one. You feed it a start and end height/layer, the number of tweaks you want it to do along that range, start and end values and voila. For now, works with extruder temperature, filament flow and fan speed (can add more easily, contact me if interested). Absolutely great for doing temp/flow/fan calibration towers. Hope you'll find it useful as I do. It also outputs a neatly formatted summary of the tweaks at the beginning of the GCode file.
→ For Cura 2 and up. (Tested on Cura 3.1 but should work ok with any version)
→ Installation : Just download the file and put it in "Program Files\Ultimaker Cura 3.1\plugins\PostProcessingPlugin\scripts" (or the corresponding folder on other OSes)
→ Instructions : See attached image.
→ Download : https://drive.google.com/open?id=1FkbCIZuUKl_EHxL_w40aQYXh-0Vjj_Yk

[image]

ChrisTerBeke, 2018-02-15:
Seems cool! If you want it to be included in Cura, please create a pull request with the code changes so we can review and merge it.

JJJpnt, 2018-02-15:
Thanks. I'll do that indeed, but even if it definitely works, before making a pull request I'll work a little on little cosmetic code works, and maybe test it a little bit more thoroughly and add a safeguard here and there (like if someone puts so much steps that the increment is finer than the layer height I have no idea yet how it would handle that). I don't know who did the original TweakAtZ plugin, big thanks to him !!!

Ghostkeeper, 2018-03-08:
Careful with those long lines in the summary. Some printers don't handle lines of more than 80 characters well. They introduce their own line breaks, causing the line to be broken up and the second line is not starting with a semicolon so it won't be commented out.

diegopradogesto, 2019-09-10:
We are looking forward to this changes. If you still plan to create a PR, I think it's better to continue the conversation there.

Feel free to reopen if you think it still needs to be here.

closed 2019-09-10 by diegopradogesto