Ultimaker/Cura #3329: Cura 3.2, Z-Hop not working.

Thisismydigitalself, 2018-02-15:
My Cura 3.2 doesn't perform Z-Hop. 
For troubleshooting this I entered a value of 9.8765 and searched generated gcode but could not find it except in the settings section: **;SETTING_3 op = 9.8765\\nretraction_hop_enabled = True\_** ...

. usually i use Zhop of 1mm and retraction minimum distance of either 1mm or 10mm for checking it's functionality. it worked up until couple of days ago. can't remember if in 3.1 or 3.2. i would have checked with 3.1 but for some reason it refuses to launch.

ChrisTerBeke, 2018-02-15:
My z-hops work fine...

Please share your user settings, without that there's no way to debug for the developers.

Thisismydigitalself, 2018-02-15:
Here you go (change .txt to .3mf)
[L_TOOTH v5.curaproject.txt](https://github.com/Ultimaker/Cura/files/1729735/L_TOOTH.v5.curaproject.txt)

smartavionics, 2018-02-18:
Turn off "z hop only over printed parts".

smartavionics, 2018-02-18:
BTW, when I have a model like yours with horizontal holes, I put a thin web across the hole so that the regions either side of the hole are always connected by outer walls. Here's a grotty image that shows where the webs are positioned. Once the print is finished you have to drill out the holes but that only takes a few moments. I find that the wall quality is improved by doing this.
[image]

Thisismydigitalself, 2018-02-18:
Turn off "z hop only over printed parts" ?  
This is counter intuitive. I left this feature checked few weeks back (and forgot about it) thinking no need to hop outside the part, what am i missing?.. :)

Can you please explain what do you mean by "thin web" across the holes?

Thx.

smartavionics, 2018-02-18:
The webs are simply a vertical wall that completely blanks the hole. I normally position them on the centre line of the holes (the orange lines in the image above) but sometimes I position the web flush with one of the part's side walls. I normally make the web 2 x the outer wall line width thick so that it is printed as part of the outer wall. The point is that the outer wall of the hole part is printed in one with no retracts/travels.

Thisismydigitalself, 2018-02-18:
> The point is that the outer wall of the hole part is printed in one with no retracts/travels.
Understood, good idea.

smartavionics, 2018-02-18:
Here's an example, there are 4 horizontal holes and each has a web on its centreline:

[image]

See how the outer wall for the whole part is printed in one.

Thisismydigitalself, 2018-02-18:
Yes, perfectly understood.

Thisismydigitalself, 2018-02-19:
After unchecking "Z hop only over printed parts" Zhop works again.
Please explain the reasoning behind the description as it is counterintuitive for me.

Thx.

smartavionics, 2018-02-19:
>  Please explain the reasoning behind the description as it is counterintuitive for me.

It's not my work, I don't know what they intended to achieve with that, sorry.

Thisismydigitalself, 2018-02-20:
[image]

smartavionics, 2018-02-20:
Looking good although are the webs too thin? They have to be 2 outer wall thicknesses wide.

Thisismydigitalself, 2018-02-20:
[image]

smartavionics, 2018-02-20:
That's looks good. Have you tried printing it?

Thisismydigitalself, 2018-02-20:
will print it later today. i placed the web closer to one end for either closer drill, or for using the longer side for making sure the drill bit will have more bore to straiten itself before it hits the web. we'll see.

ianpaschal, 2018-03-09:
@Thisismydigitalself Any luck? Can we close this?

Thisismydigitalself, 2018-03-09:
You can close this issue on one condition:  Please tell me why one needs to uncheck "Z hop only over printed parts" for things to work as expected. this somewhat counterintuitive description was what got me me confused in the first place. please consider better description :)

ianpaschal, 2018-03-09:
I can't really defend this kind of thing as I've only been working at Ultimaker 10 weeks. I could dig out the commit and figure out who named the setting but I think a better option is to tag @LukeChen-Ultimaker. Luke and I both did the same UX masters program but he's been at Ultimaker far longer than I have so he may know better about why it is the way it is. He might also feel like investigating the issue from a UX perspective. Not sure. I am going to go ahead and close the issue because we will consider it, but I have to defer to my seniors regarding what we do about it.

closed 2018-03-09 by ianpaschal

Ghostkeeper, 2018-04-04:
> After unchecking "Z hop only over printed parts" Zhop works again. Please explain the reasoning behind the description as it is counterintuitive for me.

As it seems nobody was able to explain this so far, I think we need to clarify this a lot better then!

The point of this setting is that if you're performing a z-hop from A to B but the path between A and B is completely devoid of any obstacles, then the hop is useless. You wouldn't hit anything in the path to B anyway. So if you're ending at A in an outer wall and starting at B in an outer wall, then you might as well not perform a Z hop and save some time and wear on the Z axis.