Ultimaker/Cura #3330: Cura Won't Open with Fusion 360

kenbo017, 2018-02-15:
I installed Cura 3.2.0.  I am also using Fusion 360.  I set Cura as the application for printing 3D parts.  When it goes to open Cura the Cura program opens and then disappears very quickly.  Then I'm no longer able to open the program on its own without going through Fusion but directly to open Cura.  When I remove the program and reinstall it will open directly again but if I try through Fusion 360 it no longer stays open either through Fusion or directly again.  It does the same opening but closing immediately.

Now I removed and reinstalled again.  It will no longer stay open even when I tried to open only directly without ever trying Fusion 360.  I have seen plenty of people have this issue but haven't seen a resolution.

alekseisasin, 2018-02-16:
Can you share the log file.

kenbo017, 2018-02-16:
I can't find a .log file in the C:\Program Files\Ultimaker Cura 3.2 folder or any of its subfolders.

kenbo017, 2018-02-18:
I installed the latest version 3.2.1.  I'm still having this issue of it crashing just after opening through Fusion 360.

alekseisasin, 2018-02-19:
I think the problem is not related to the Fusion 360. The log file you can find:

The Cura GUI log file, located at
    %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
    $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
    $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

alekseisasin, 2018-02-19:
did you use before earlier Cura version. If yes, then you can try by yourself remove Cura user-settings. Because might be after upgrading the Cura version something went wrong. Then you will need to remove 
%APPDATA%\cura ( C:\Users\\AppData\Roaming\cura).

kenbo017, 2018-02-20:
> Hello,
>
> Thank you for the follow-up!  I really need to get this figured out.  I
> have not used Cura before my install of 3.2.0.  As I mentioned, I did
> install your latest update that said it fixed the crashing on startup.
>
> Attached is my log file.
>
> Please let me know what you find.
>
> Regards,
>
> KS
>
> Ken Schneider
> Schneider Sales and Marketing
> 312 West Hillcrest Blvd.
> Inglewood, CA 90301
> ph: 310-680-9544 <(310)%20680-9544>
> cell: 310-722-8827 <(310)%20722-8827>
> fax: 310-861-5798 <(310)%20861-5798>
> e-mail: kens@ssmcentral.com
>
>
> CONFIDENTIALITY NOTICE. This electronic mail transmission may contain
> privileged, confidential and/or protected information and is intended only
> for the review of the party to whom it is addressed. Any unauthorized use
> or disclosure of the information contained herein may be a violation of
> federal law. If you have received this transmission in error, please
> immediately return it to the sender, delete it and destroy it without
> reading it. Unintended transmission shall not constitute the waiver of the
> attorney-client or any other privilege.

Hello,

I got the message bounced back because my log file is too large  It's 5.5
megs.  I've saved it into my google drive so you can download it.  Please
see my comments below from my original reply.​
 cura.log
<https://drive.google.com/a/ssmcentral.com/file/d/1UnpnHWYwhul3I_t8PUtHzALAOwKfw2D7/view?usp=drive_web>
​

Regards,

KS

Ken Schneider
Schneider Sales and Marketing
312 West Hillcrest Blvd.
Inglewood, CA 90301
ph: 310-680-9544
cell: 310-722-8827
fax: 310-861-5798
e-mail: kens@ssmcentral.com

CONFIDENTIALITY NOTICE. This electronic mail transmission may contain
privileged, confidential and/or protected information and is intended only
for the review of the party to whom it is addressed. Any unauthorized use
or disclosure of the information contained herein may be a violation of
federal law. If you have received this transmission in error, please
immediately return it to the sender, delete it and destroy it without
reading it. Unintended transmission shall not constitute the waiver of the
attorney-client or any other privilege.

alekseisasin, 2018-02-20:
@kenbo017 Can you give me access to the file. I sent a request.

alekseisasin, 2018-02-21:
I see in log file is CRITICAL error. We will try to investigate this case and find the reason of it.
DEV: CURA-4989

alekseisasin, 2018-02-21:
It looks like that error occurs because of OpenGL, do you have right drivers for your video card. Even sometimes the latest drivers have issue. Can you try to install latest or earlier video card drivers.

On my PC the Cura works fine with Fusion 360.

kenbo017, 2018-02-21:
I updated my video driver and it seems to be working perfectly now.  I'm so
excited!

Thanks a ton for your help.  I was going crazy removing and reinstalling
Cura all the time just to be able to use it.  I'm new to the 3D printing.
I just got an JGAurora A5 printer.  I've got it pretty dialed in now with
the setting so I can start trying some real prints so I'm very excited.  It
seems like this printer (fairly new to the market) is becoming pretty
popular as an entry level printer.  Perhaps you might want to add it to
your list of printers.  I'd be happy to give you any feedback I can so you
can add it to your list.

Regards,

KS

Ken Schneider
Schneider Sales and Marketing
312 West Hillcrest Blvd.
Inglewood, CA 90301
ph: 310-680-9544
cell: 310-722-8827
fax: 310-861-5798
e-mail: kens@ssmcentral.com

CONFIDENTIALITY NOTICE. This electronic mail transmission may contain
privileged, confidential and/or protected information and is intended only
for the review of the party to whom it is addressed. Any unauthorized use
or disclosure of the information contained herein may be a violation of
federal law. If you have received this transmission in error, please
immediately return it to the sender, delete it and destroy it without
reading it. Unintended transmission shall not constitute the waiver of the
attorney-client or any other privilege.

alekseisasin, 2018-02-22:
Good to know that we helped you. Welcome to the 3D printers community.
We are open to add Third-party printers like JGAurora, only they should to make a Pull Request and provide printer definition files.

closed 2018-02-22 by alekseisasin

kenbo017, 2018-02-23:
Thanks!

Do I need to do something to have them create a pull request?  As I'm new
to this I have no idea what to do.

Regards,

KS

Ken Schneider
Schneider Sales and Marketing
312 West Hillcrest Blvd.
Inglewood, CA 90301
ph: 310-680-9544
cell: 310-722-8827
fax: 310-861-5798
e-mail: kens@ssmcentral.com

CONFIDENTIALITY NOTICE. This electronic mail transmission may contain
privileged, confidential and/or protected information and is intended only
for the review of the party to whom it is addressed. Any unauthorized use
or disclosure of the information contained herein may be a violation of
federal law. If you have received this transmission in error, please
immediately return it to the sender, delete it and destroy it without
reading it. Unintended transmission shall not constitute the waiver of the
attorney-client or any other privilege.

ianpaschal, 2018-02-23:
You should ask them to add their printers to Cura and send them the repository (and a link to this issue if you like, although your original issue is not really related).

Hopefully they will find time to write the code, and can then make a PR (pull request). A pull request is a request to us to "pull" their changes to the code into the official version.