Ultimaker/Cura #3331: Janky Input fields with Auto Slice enabled

SaltyLiquorice, 2018-02-16:
**Application Version**
3.2 - 3.2.1

**Platform**
Win10

**Steps to Reproduce**
Enable Auto Slicing and try to change settings values with layer view.
Less pronounced in solid or xray view.

**Actual Results**
Slicing occurs during input interrupting the user. The cursor is almost always removed from the setting field as well forcing the user to click back in.

**Expected results**
Slicing should wait till the user has hit enter or changed to another input field.  At least until a whole 3 digits has been entered.

**Additional Information**
I saw there was another issue #2947 posted a while back with a post claiming 500ms was the wait time after user input before auto slicing.  It feels like this might be broken at the moment.

alekseisasin, 2018-02-16:
@SaltyLiquorice  The same setting input is shared between all settings and some of them accept minimum one digit (like Support Speed). For this reason "at least a until whole 3 digits" cannot be a constraint. 
You mentioned start slicing after pressing "Enter". It can be, but also you can see each time when you change the value (like Outer Wall Line Width) other settings are automatically recalculated. Then every user should know if he wants to start slicing he should hit _Enter_ . 
Did you try to disable auto slicing. If you will open _Preferences_ - > Disable _Slicing automatically_.

SaltyLiquorice, 2018-02-16:
Thanks I hadn't considered those fields.  Yeah I did disable it but it is pretty handy to have when it works well and did work well before 3.2 in all view modes.

alekseisasin, 2018-02-16:
What do you mean "did work well before 3.2", Maybe it triggers slicing much often then before because we try to enhance the cura speed. If you can share you ideas then I would appreciate.
According to you "Input fields with Auto Slice enabled". For now we do not have any clear scenario which would fulfilled your request. Maybe after moving to a new version of Pyqml we will come to a new elegant solution.
For now If you do not mind I would like to close the ticket.

SaltyLiquorice, 2018-02-16:
Sorry if I wasn't clear I was in a rush. Before Cura 3.2 Auto Slice had a much larger delay after input was completed.  I could type my 0.25 layer height and it would slice regularly after that input.  Now if I try the same input it will slice after input of the digit 0 before I can add the .25.  After the slice is complete the cursor is no longer in that input field.  I have to click back into the input field and add the next digit.  This may not even happen for every input but it happens frequently enough to be frustrating and for anyone who has the feature enabled to turn it off because of that frustration.  Auto slice is enabled by default on installing Cura? If this is the case and people use layer view they will all be having this problem.

I don't have any ideas on how to fix this I just thought I should bring it to someones attention since I couldn't find much info on it.

fieldOfView, 2018-02-16:
For the record, I have auto-slicing enabled and I don't get the problem that slicing starts immediately after entering a digit. What I do see however is that every time slicing completes, the layer slider grabs the focus away from the sidebar, which I agree is quite annoying.

Ghostkeeper, 2018-03-08:
The slicing should be waiting 500ms before it starts.

We've changed a few things regarding this in 3.3 so maybe this will be somewhat better by the next release.

ianpaschal, 2018-03-09:
Since we've improved this in 3.3, I'm marking it as fixed. If it's still an issue we can re-open it.

closed 2018-03-09 by ianpaschal