Ultimaker/Cura #3332: Bed Temp Not Shown in Monitor

Nuklear47, 2018-02-16:
**Application Version**
3.2.1 and earlier (3.0.4 works ok)

**Platform**
Older Hictop I3 Mk2

**Display Driver**
NVIDIA GeForce 710

**Steps to Reproduce**
whenever Cura 3.2.1 or earlier is opened (3.0.4 works ok) you go preheat the bed and the current temp
is shown as 0 degrees even though the temp is ambient or current bed temp.

**Expected results**
This should be self explanitory...

**Additional Information**
3.0.4 worked ok originally then i upgraded to 3.1 and lost bed temp info in the monitor screen

slundell, 2018-02-21:
Are you printing over USB/Serial?

Nuklear47, 2018-02-22:
Usb

<https://github.com/Ultimaker/Cura/issues/3332#issuecomment-367339478>, or mute
the thread

alekseisasin, 2018-06-29:
Hi @Nuklear47, did you succeed to send manually M105 command. Or do you still have the same issue in the latest Cura ?

Nuklear47, 2018-06-29:
It was just cura being glitch, now that they have updated it works again

diegopradogesto, 2018-07-24:
Good to hear! I will close the issue then.

Which version are you using now, that you say it works? Is it 3.4.1?

closed 2018-07-24 by diegopradogesto