Ultimaker/Cura #3334: Cura 3.2.1 freeezes when slicing

abj80, 2018-02-16:
**Application Version**
3.2.1

**Platform**
Windows 10 64 bit

**Steps to Reproduce**
After loading a model, pressing the "Prepare" button freezes Cura. I even did a clean install (removed all the files related to older Cura versions) but I still get the same result, even with the default settings.

I have attached the log file.

Thank you!

[cura.log](https://github.com/Ultimaker/Cura/files/1730902/cura.log)

smartavionics, 2018-02-16:
Hi, I don't know what is the cause but in your log file you have the following error:
````
2018-02-16 12:11:29,367 - DEBUG - [(10236)-Thread-12] UM.Backend.Backend._backendLog [90]: [Backend] [ERROR] Arcus Fatal Error (2, native 10013): Could not connect to the given address
2018-02-16 12:11:29,431 - DEBUG - [(7376)-MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [722]: Backend quit with return code 0. Resetting process and socket.
````

Hopefully, an Ultimaker person can advise you as to what is wrong.

abj80, 2018-02-16:
It seems that the culprit was the firewall. After some adjustments, everything works fine now.

smartavionics, 2018-02-16:
Great, please close the issue if everything is now OK. Cheers.

closed 2018-02-16 by abj80