Ultimaker/Cura #3335: Cura 3.1: Z-seam inner walls do not match z-seam outer wall

ozfunghi, 2018-02-16:
**Application Version**
3.1

**Platform**
Windows

**Steps to Reproduce**
Set a model with Z-seam and layer start coordinates at the back, multiple walls and low (or gradual) infill.

**Actual Results**
Cura will indeed start a new layer at the back, it will indeed start the OUTER wall at the back, but it will start the INNER walls at the position where the infill happens to finish every layer. Thus creating an aditional seam, at the place where the INNER walls start and stop, before moving to the back for the OUTER wall.

**Expected results**
Z-seam coordinates for both outer AS WELL AS inner walls. One would think this would have been obvious, since any travel to these coordinates had to be made sooner or later and since the length of outer and inner walls are basically the same and cover the same perimeter anyway.

**Additional Information**
Video (zipfile) starts when the outer wall starts, lasts longer than the entire duration of one layer. So inner walls and infill can be observed as well. [I had already placed a somewhat related issue some months ago](https://github.com/Ultimaker/Cura/issues/2458), where the layer start coordinates had to match the z-seam coordinates to avoid nasty seams, but it's not the same as this.

It's is really frustrating, that basic stuff such as this is still messing up prints. I even use the (wonderful) fuzzy skin option to cover any inconsistancies in layers or zits, but even that isn't enough to cover up the marks made by the inner wall z-seam
[cura-seam.zip](https://github.com/Ultimaker/Cura/files/1730898/cura-seam.zip)

smartavionics, 2018-02-16:
Have you tried using the optimize wall print order setting? That should align the z-seam of the inner walls with the outer wall z-seam.

ozfunghi, 2018-02-16:
Yes, the print in the video has optimize wall print order enabled.

smartavionics, 2018-02-16:
OK, in that case, if possible, please make the gcode available for me to see what's happening.

ozfunghi, 2018-02-16:
Here you can see a screenshot with the settings (z-seam + optimize wall printing order) and you can see the position where the printhead starts its inner wall after the infill has finished.
[image]

Do you want me to upload the gcode as well?

smartavionics, 2018-02-16:
Yes, please, I need the gcode so I can step through it.

ozfunghi, 2018-02-16:
[z-seam-issue-inner-walls_bender-ozfunghi-v3.zip](https://github.com/Ultimaker/Cura/files/1731100/z-seam-issue-inner-walls_bender-ozfunghi-v3.zip)
I zipped it (it was 55MB otherwise).

ozfunghi, 2018-02-16:
PS: this gcode was made with cura 3.2.1 (i installed this last night) but as the layer preview shows, it also has the same issue as 3.1 before it. But that gcode is currently printing and on an SD card so i had to create the gcode again.

ozfunghi, 2018-02-16:
Thanks. If "infill before walls" is disabled, the preview shows both inner and outer walls do start at the back (though in some cases it starts really at the back, and in other layers at the position where an infill ended... "almost" at the back). But this is a model where i'd like to keep that option enabled (infill before walls). And regardless, i think it should still start inner and outer walls at the same spot. 

This is with infill before walls disabled: You can see one layer starts at the back, another not at the back but where an infill ends.
[image]

smartavionics, 2018-02-16:
Can you please make the STL available? Although I can see the problem occurring in the gcode, I think I need to slice it myself to really understand what's going wrong. It's definitely not right at the moment.

ozfunghi, 2018-02-16:
https://www.thingiverse.com/download:4078211

smartavionics, 2018-02-16:
Just had a thought, if you turn off the fuzzy walls, do the z-seams become
aligned?

ozfunghi, 2018-02-16:
Not according to the layer preview in Cura. Problem still remains with fuzzy skin turned off.

ozfunghi, 2018-02-16:
I just recreated the same issue with a completely different model as well.

ozfunghi, 2018-02-16:
Also with gradual infill turned off and a constant infill, the problem remains.

smartavionics, 2018-02-16:
Hi, I think I know where the problem lies (it's a combination of infill first and inner walls first that causes the inner walls to not start near the outer wall). I think I have a fix that can go into a future release but I want to do some further testing over the weekend. Sorry for the inconvenience.

ozfunghi, 2018-02-16:
I'm pretty sure the problem lies with the optimize wall printing order. As soon as i turn that OFF, the problem goes away. Ironically.

ozfunghi, 2018-02-16:
[image]
The only thing changed, is the optimize wall printing order setting... and here it prints as it should.

smartavionics, 2018-02-17:
The problem you are seeing is triggered by the combination of infill before walls, inner before outer walls and no holes in the part. I have submitted a fix (https://github.com/Ultimaker/CuraEngine/pull/705) which just disables the optimization when the part has no holes so it falls back on the non-optimizing code which now (it didn't used to) aligns the z-seams of the inner and outer walls.

closed 2018-03-09 by Ghostkeeper