Ultimaker/Cura #3336: Material duplicates disappear when changing diameter and <TAB>

povlhp, 2018-02-16:
Cura 3.2.1 (also had it in 3.0 but never reported).

I have EU locale, that is decimal comma.

To get the issue, duplicate any material of generic brand. Say ABS.

Set diameter to say 5,0mm - tab out of input box, and material disappear.

Now, select the master "unmodifiable" "generic" ABS again, and it is listed as having 5mm diameter.

Now, to actually get a working clone of the material, you can enter a value of say 1,76mm and press down arrow to the right to decrease diamater to 1.74mm. And your duplicate material will stay. You can then open it again, and enter values and tab, and it will still not disappear.

So it happens only on the first edit of a duplicate material.

It is repeatable, and I run into it often when creating a new material - Different brands need different extrusion, and easier to adjust diameter on a filament brand/color than a print profile per filament.

nallath, 2018-02-20:
This is semi-intended. Materials are filtered by the diameter that the machine can handle. It does round a bit, which is why setting it to 1.74 does allow it to remain visible, but setting it to 5.0 means it's not.

fieldOfView, 2018-02-20:
It should not disappear without a message though.

povlhp, 2018-02-20:
But there are multiple issues. One is, that when I change the copy, that disappears, then the master gets the new invalid diameter. 
And TAB should not be seen as a final acceptance of the values, it should jump to the next input field as per standard on most platforms I have used.
And rather than removing/hiding the material, it would be better with an alert.

mgalvey, 2018-02-21:
I just spent about 20 minutes trying to figure out why this was happening. To the end user, it definitely looks more like a silent failure than intended behavior. My suggestion: add a checkbox in the materials management screen for "Only show materials compatible with selected printer" or something along those lines. At the very least, some kind of prominent note should be visible on that screen explaining this functionality.

povlhp, 2018-02-21:
But the note will not fix the illegal value getting copied back up to the master material. That alone is a bug as I see it.
If I create a new material from the master, it starts out with the illegal value. 

It would be better to always have the diameter of the master material have a compatible diameter with the current printer. If this was the case I would not have had the issue. Then I would not have had to change the diameter. Only had to do it in the first place as my custom printer is 1.75mm, and default materials was 3mm.

damienraven, 2018-02-28:
OK... I see that the material that shows up is only the ones that are consistent with the diameter that the machine can handle, but I have two questions:
- Why all 2,85mm materials shows up all the time, with a 1,75mm machine?
- Why all the 1,75mm materials don't show up on a 1,75mm machine? (I set the machine to 1,76mm and then they all showed up)
I think that the suggestion made by @mgalvey (this would be awesome) and a warning message if a inconsistent diameter selection is made (i.e. material 2,85mm chosen on a 1,75mm machine and vice versa) will solve this "problem".

Ghostkeeper, 2018-03-23:
> And rather than removing/hiding the material, it would be better with an alert.

That should be happening. Expected behaviour is that the user is allowed to change the filament diameter. When the approximate diameter (rounded to integer millimetres) is different from the approximate diameter that the nozzle supports, then a message should pop up asking if you're sure you want to make this change.
[image]

> And TAB should not be seen as a final acceptance of the values, it should jump to the next input field as per standard on most platforms I have used.

Cura has always held that changes are applied immediately, no apply buttons or anything. It's a bit of a style thing, but at least on my operating system (Ubuntu) this seems to be the standard. On Windows from what I've seen the operating system itself has these "apply" buttons but in the applications it varies from program to program.

> My suggestion: add a checkbox in the materials management screen for "Only show materials compatible with selected printer" or something along those lines.

All the generic profiles currently have two versions, one for 2.85mm and one for 1.75mm diameter. We gave these profiles the same user-visible name. So if you were to show all materials instead of just the currently supported ones, you'd see every base material twice.

> But the note will not fix the illegal value getting copied back up to the master material. That alone is a bug as I see it.

I cannot reproduce this. Please create a new bug report if you still see it happening with some nice reproduce steps.

> Why all 2,85mm materials shows up all the time, with a 1,75mm machine?

I also cannot reproduce this. Reproduce steps? Could you post this in a new bug report?

> Why all the 1,75mm materials don't show up on a 1,75mm machine? (I set the machine to 1,76mm and then they all showed up)

Are you sure that the printer hadn't set the supported material diameter to 2.85 instead of 1.75? Or were you seeing no materials at all or something? This could also be worth a new bug report.

This thread is about the warning displayed when you change the material diameter to something else than what the printer supports. Let's keep it concise.

danfoster, 2018-04-07:
Just wanted to mention I'm seeing similar behaviour. I created a custom printer which had supported material diameter of 1.75 by default. When I created a material of 1.75mm diameter, it would disappear from view.
I went back to my printer settings, changed the supported material diameter away from 1.75mm (e.g. 1.76mm), then back to 1.75mm. When I went back to the material list, all my 1.75mm diameter material were visible.

Could it be possible that the "default" settings for a custom printer is 2.85mm even though the UI displays 1.75mm?

Ghostkeeper, 2018-05-01:
Which printer was that?

ianpaschal, 2018-05-29:
Needs more info but has not received it for almost two months. The case has gone cold. :(

closed 2018-05-29 by ianpaschal