Ultimaker/Cura #3337: Cura 3.1: Z-hop causing consistently missed layers. Conflict with another setting?

fmvjack, 2018-02-16:
**Application Version**
Cura 3.1

**Platform**
Windows 7

**Issue**
With the attached config ([cura config.txt](https://github.com/Ultimaker/Cura/files/1731202/cura.config.txt)), If I ever turn on z-hop, I get what look like entire missed layers. 
You can see from the pictures attached of the calibration cube. It has happened with any model I've printed with z-hop on, and always in the same place per model. i.e a calibration cube will always be on layers 'a' and 'c' whereas a temp-tower will always miss a layer at layers 'x' and 'y'.

I'm guessing that it's conflicting with some other setting in the slicer ?

But the issue is plain as day to replicate. Turn on z-hop, issue appears. Turn off z-hop, it goes away and prints ok.

This picture (https://i.imgur.com/peXPTyD.jpg) 

shows the issue. the left three prints have it turned on. the right way another print to show that its not a mechanical issue at that height etc. I scaled the 3rd calibration cube, and the issue followed the part of the print, so again shows it's not mechanical.

smartavionics, 2018-02-18:
Hi, could you please make the gcode that has the problem available for me to look at? Thanks.

rgrae81, 2018-02-27:
Had the same Problem on cura 3.2.1 on WIN10

Identified the Problem with Z-Hop (retraction-hop) in combination with combining.
If z-hop is on and combining is set to all or noskin, then there is a layer issue.

no issues when z-hop is off and combining is on or off
or
when z-hop is on and combining is off.

Pictures from print and a compared gcode view. left z-hop is on, right z-hop off both wich combining: noskin

[image]
[image]

smartavionics, 2018-02-27:
Hi @rgrae81 , could you please make the bad gcode available so I can study it? Thanks.

rgrae81, 2018-02-28:
Hi, I created another Thing (half of an Egg with holes) it's smaller :-) and the problems are also visible.
But I can upload the cat gcode too if you really need.
[egghalf.zip](https://github.com/Ultimaker/Cura/files/1765843/egghalf.zip)

smartavionics, 2018-02-28:
Hi, thanks for the egg gcode and pics. Unfortunately, I can't see anything wrong with egg_zhop_combnoskin when I look at it in a gcode viewer. Could you please provide the gcode for the bad animal(!) and maybe I can see the problem in that. Thanks.

[image]

rgrae81, 2018-02-28:
HI, 
But it's definitly there, I printet two versions of the egg to compare
[image]

some more Information may be helpfull:
Printer is CR-10s with Marlin 1.1.8
I used ideaMaker to view the GCode Files.

Animal-Gcode will follow

rgrae81, 2018-02-28:
Here's the cat-gcode, with default draft profile only z-hop and coming modyfied.
[catstretch.zip](https://github.com/Ultimaker/Cura/files/1767101/catstretch.zip)

I also revied the G-Codes from the egg, it seems that it has something to do with Extrusion-Amount and Retract.

Also in the Viewer the Problem is gone if I set the Option "Retract Before Outer Wall" to enabled, I'll test this later today on a Print.

smartavionics, 2018-02-28:
I've looked at the cat gcode and cannot see the horizontal lines that are visible in the left hand gcode view in your previous comment. Is catstretch_zhop_comball.gcode really exactly the same file as in the above image?

rgrae81, 2018-03-01:
Please open it with ideMaker or simple print it.
Please look at my egg-prints, the lines are there.
I also found out, that it has nothing to to with the coordinates, It seems that it have to do with retraction and extruding amount. If you disable zhop at the egg at Layer 4 it also don't retract, but as i understand the function z-hop it only do it at retraction. And also the Amount of extruding is different at that point. Use notepad++ to review.

I did other G-Code-Files and if you enable relative-extruding the extruding amounts are the same
If you enable Retract Before Outer Wall the problem is also gone.

Please don't hang up on gcode-viewer-viewing, don't know how strong the all the viewer's interprete the extruded amount

smartavionics, 2018-03-01:
I found ideamaker and using that can see the same as your picture above. However, 3 other gcode viewers (cura, s3d and craftware) all display the gcode without those horizontal lines. So my guess is that ideamaker has a problem. Also, looking at the gcode, I cannot (yet) see any obvious errors. Sorry, I don't what the issue is here but I am not saying that cura is OK, just that I do not yet understand what's causing your bad print.

smartavionics, 2018-03-01:
BTW, I looked at your config values and you have a retraction amount of 0.05. Assuming that value is in mm, that's mighty little for a bowden extruder. IMHO, totally inadequate.

rgrae81, 2018-03-01:
Hi,
you saw the lines in ideamaker and please compare it to my prints.
btw. the retraction amount is 5mm at 90mm/s this 0.005 is the retraction extra prime amount.
I did the egg-print now sliced by s3d, there are no problems at all

Sorry, I can't do more than tell you the problems, show you the prints provide you the files.
Also I helped others with the same problem with telling them to deactivate z-hop or combing, and this worked for them also. 

To say, Cura works fine cause the most slicers dont show the Problems and ignore that there is a problem you can see on a print is an option for sure.
But is this a solution?

This is an Open-Source-Software, I'm using it, and I would like to give back a bit, by helping to find bugs, and also helping to find a soloution. If this is not desired, I'm helpless.

If you want to print something here are the STL-Files!

STL for the Cat:
https://www.thingiverse.com/thing:1565405
STL for the Egg is attatched
[egg-test.zip](https://github.com/Ultimaker/Cura/files/1771420/egg-test.zip)

smartavionics, 2018-03-01:
Your config has this line in it:

> retraction_amount = 0.05

That is 0.05 of a mm so you are really not getting any retraction at all.

I understand your frustration and appreciate your willingness to help solve the issue but, right now, I cannot see where the problem is coming from. I am not ignoring you or saying you are wrong.

rgrae81, 2018-03-01:
you looked at the config from fmvjack (Threadstarter)
That's not me! (rgrae81)

I have 
retraction_amount = 5
retraction_extra_prime_amount = 0.005
retraction_hop = 0.5

rgrae81, 2018-03-01:
and I testet the extra_prime-amount from 0.1 to 0.005 to find the best value witch is working for my printer and my bowden.

smartavionics, 2018-03-01:
OK, sorry, my mistake. I am currently printing the egg with combing = no skin and z-hop enabled and it looks fine. It is doing walls before infill so I will do another with infill before walls to be sure.

rgrae81, 2018-03-01:
I printed infill before walls. Maybe you can give me your gcode of the print so I can compare it to mine?

smartavionics, 2018-03-01:
Here's the gcode I just printed, it's sliced for my Kossel delta.

[KMMB_egg-test.zip](https://github.com/Ultimaker/Cura/files/1771604/KMMB_egg-test.zip)

The prints came out fine apart from near the top of the egg where there are gaps because the infill is not dense enough. The only horizontal line visible is aligned with the top of the holes.

rgrae81, 2018-03-01:
I sliced the egg now again, with fresh installed cura, and did not import my profiles. take the standard draft and changed only z-hop, combing and relative extrusion, for better view at comparing both gcodes. Can you explane me the differences, I thought enabling z-hop will only do a hop. This happens not on each retract, but nearly
[image]

rgrae81, 2018-03-01:
I opened your gcode in ideamaker and also can see a Line direct above the holes! You mean that, or you have a line in your print above the holes?

rgrae81, 2018-03-01:
> I am not ignoring you or saying you are wrong.

Sorry about that, I missreaded your post above about NOT OK,

smartavionics, 2018-03-01:
Yes, there is a visible line above the holes but that I think is due to the fact that I am using some experimental bridging settings that modify the speed/flow/fan in the bridge region and this has an effect on the wall around the whole object. If you look closely at the images below you can see that just before the bridge lines occur, there's a little bit of coasting (no extrusion) and the bridge lines themselves are printed slower and thinner (and the fan goes faster).

[image]
[image]

However, ignoring that, there are no poor horizontal lines like you have in your image.

rgrae81, 2018-03-01:
Can you please provide me a GCode from the egg with following options:
Printer CR10
Default PLA
Default Draft 0.2mm profile
z hop on.
combing no skin
Everything else default. I'd like to compare our both gcodes with this settings. 
Cause this both files should be the same i think, and I'd like to check this.

smartavionics, 2018-03-01:
Sorry, I run a version of Cura that has some differences to the standard releases so the gcode I would produce with exactly the same settings would not be identical to what you would produce.

rgrae81, 2018-03-01:
Ok, this Info is interesting. Maybe I also don't have Prolems with this Version? Could you provide me this version so i can print on this one with my profile?
btw. could you take a look at the gcode I posted above?

smartavionics, 2018-03-01:
Sorry, I cannot provide you with my Cura because I can't build a release. I just build it from source on my Linux box. The source code is freely available, it's not secret, I just can't build for anyone else.

As for the gcode samples above, the one on the left is retracting, doing a z-hop, travelling directly to the destination, moving up by a layer height (which is actually not needed because the z-hop has already raised the head) and then it's lowering to the height of the next layer and doing a prime (aka un-retract).

The one on the right is doing the retract, then it looks to be combing to the destination where it raises the head to the next layer height and then it does the prime.

rgrae81, 2018-03-01:
OK, so you use an selfcompiled cura with some differences to the normal one on linux with an delta Printer, and I use the 3.2.1 release on Windows10 on a cartesian printer.

Maybe this Problem only affects cartesian Printers, maybe only with Marlin 1.1.8, maybe only Windows-installations. Don't know. Do you know some contributors wich have a cartesian printer and running 3.2.1 on Win10, maybe they can reproduce it?

fmvjack, 2018-03-05:
Sorry guys, completely forgot to come back and provide examples.

I'll run off two sets of gcode later for the calibration cube, one with , one without z-hop turned on.

I'll also check out ideamaker and see if that makes the issue visible before printing.

fmvjack, 2018-03-05:
Yeah, I also tried a different slicer. It is definitely something weird going on in cura :(

They  have released a newer version than mine now, so I might try and update it later to see if it fixes my issue. do you know if settings/profiles are saved on updating?

moeff, 2018-03-07:
Hi, i have the same issue with z-hop on, combining no skin: missing layers. Same model every z-height, other model other z-height.

z-hop off all is perfect. slic3r does the job perfectly with z-hop/lift. So there is no hardware issue on dirty z-axis.

printer: cr-10
cura: 3.2.1 on WIN7

seems turning back after z-lift istnt calculated correctly...

Ghostkeeper, 2018-03-22:
Turning back is done by putting the same Z coordinate as the rest of the layer in the g-code.

Since we seem to be unable to reproduce this by just looking at layer view or printing with our own printers, I think we'd need a snippet of g-code that is wrong to be able to debug this.

ianpaschal, 2018-05-29:
Needs info but hasn't gotten it for 2+ months.

mooklepticon, 2019-03-16:
I'm still getting this. I can repeat it. I've made ~15 prints with this issue before I found this page.  I discovered it when printing cal cubes of different size. It's not just the Z hop, it's the Z hop combined with the top layer of another object. 

My print was 3 cal cubes: 75%, 100%, 125%. The 75% and the 100% were fine, but the top layer of the 100% affected the 125%. Where ever the 100% top started, there was a skipped layer in the 125%. It's not the top of the 100% cube. It's the first of the top layers, so with N layers, it's N-3 layer that seems to cause a skipped layer on the 125% cube.  

I analyzed the g code using gcode.ws and there doesn't seem to be anything weird about it, though I admit I'm not a very experienced printer.  I asked the top youtubers and none had seen it. I asked Reddit and no one had any suggestions other than try another slicer.  I forgot where I saw it, but a side by side of Cura and slic3r seemed to show that slic3r didn't cause it.  (I've been troubleshooting this for 3 weeks.)

I have screen shots, g code, etc.   Ask me for anything; I want to help.

Ghostkeeper, 2019-03-19:
Maybe you can send us the g-code and indicate whereabouts in the print it is failing?

AbeFM, 2019-11-05:
I've been getting missing lines with z-hop but I have enough questionable other effects going on I assumed it was in hardware.

I will look into this a bit more - I massively turned down speeds and thought it looked better but I never really understood why the effect first appeared.

SwiftNinja, 2019-12-22:
For those experiencing this problem, I believe it has to do with binding on the Z-axis in a lead screw design. 

Try reducing your Z velocity to 3 in the firmware:
    #define DEFAULT_MAX_FEEDRATE {300, 300, **3**, 25}

Also ensure your max limits are factors of that:
    #define HOMING_FEEDRATE_Z 180
    #define MANUAL_FEEDRATE {2340, 2340, 240, 60.0}
    #define Z_PROBE_SPEED_FAST 300
    #define Z_PROBE_SPEED_SLOW 60

The values you use may depend on your printer, so this advice is offered entirely as 'use at your own risk'.

I had this problem when it's set to 5 - dropping it to 3 resolves the problem (for me). What I'm finding is that lead screws are surprisingly finicky, for a few reasons - minor variations in angle from top to bottom, and, even moreso, minor variations in the mounting of the Z-Lead-Screw Nut causing the nut to torque on a slight angle and change the resistance / slop on moves.

SwiftNinja, 2019-12-22:
Confirmed (for me, at least) - This issue is two things (both hardware). Really outside of the scope of Cura, but since this bug was logged here, for me, it was:

A) Binding on the Dual Z-axis on the Lead Screw Nuts due to uneven torque applied to nut attachment screws

B) TMC2130 steppers and Stealth Chop on Z. I believe that StealthChop reduces the overall available torque on the motors due to the small step increments (which is how it's "silent"). Reducing the torque available on Z for a dual lead screw design - not such a hot idea, it seems.

Nothing to do with Cura. Symptomology was the same as the ticket, for me - it looked like it was "missing layers".

Hope that helps someone.

Ghostkeeper, 2019-12-23:
Thanks for the investigation SwiftNinja. We'll assume that this was the case for the original poster as well. Chances are pretty slim anyway that we'll get the g-code file we asked for in order to investigate this. Hopefully your investigation provides some people with a workaround in hardware and firmware.

In general, stepper drivers are quite often a balance between silence and power. You can have one or the other, not both.

closed 2019-12-23 by Ghostkeeper