Ultimaker/Cura #3338: Does not remember the size of the nozzle.

SuraTech, 2018-02-16:
**Application Version**
CURA 3.2.1

**Platform**
WINDOWS 7

Does not remember the size of the nozzle.
Always 0.35

Vandrasc, 2018-02-19:
Hi @SuraTech, could you please use the bug template for reporting this? We will be able to help if we will receive more information. What printer are you using? Single or dual extrusion? 
Thank you

Thisismydigitalself, 2018-02-22:
Same here.

 

**Application Version**
3.2.1
**Platform**
Win10/64
**Qt**
Default
**Display Driver**
NV 388.13
**Steps to Reproduce**
Just found out nozzle size was set to 0.5 even thou was set by me to 0.4.
**Actual Results**
needed to edit machine settings again.
**Expected results**
Printer settings should keep settings.

Goury, 2018-03-03:
There is a lot of info in #3378
So mark this as bug already and fix is some day please

SuraTech, 2018-03-03:
guys I still could not solve the problem .... I still have the size of the nozzle 0.35 ... but I use the printer "KOSSEL PRO" ... if you select "СUSTOM" will also be 0.5

Ghostkeeper, 2018-03-23:
I could reproduce this with the following steps in current master:
* Clean installation of Cura.
* Upon first start, add Kossel Pro printer.
* Go into the machine settings and under Extruder, set the nozzle size to 0.4mm.
* Restart Cura.
* Go into the machine settings and check the nozzle size. It still lists as 0.35mm instead of 0.4mm.

However these steps work as expected:
* Clean installation of Cura.
* Upon first start, add Kossel Pro printer.
* Go into the machine settings and under Extruder, set the nozzle size to *0.399mm*.
* Restart Cura.
* Go into the machine settings and check the nozzle size. It now properly lists it as 0.399mm.

So it seems that something is going on when you set the nozzle size to what happens to be FDMPrinter's default.

Ghostkeeper, 2018-03-23:
Got it. The check against the default value was missing the index that the setting it would be stored in. Then it thought that the setting was already equal to the default (0.4mm) and it didn't need to store it.

Fixed here: https://github.com/Ultimaker/Uranium/commit/3f06e85cb79f3ceecff987c47fd8a0653c173935

Ghostkeeper, 2018-04-06:
The fix that I made here broke inheritance when typing too fast. I'm reverting it and adding a new issue so we can investigate it a bit more thoroughly.

smartavionics, 2018-04-06:
I think this has cured the problem, thanks.

Ghostkeeper, 2018-04-30:
Yeah, confirmed that it's now fixed and the settings reverting bug is fixed as well.

closed 2018-04-30 by Ghostkeeper

maverickccie, 2018-05-07:
I am have the same issue.

Printer: RostockMax v3.2
Application Version: 3.3.1
OS: Win10/64

I set the nozzle to .4 close the application. When I reopen the application the nozzle size is set to .5 again!

yuyat, 2018-05-07:
Same problems on Windows and Linux.
Printer: DeltaBot.

fieldOfView, 2018-05-08:
As a workaround, try setting the nozzle size to 0.401 or something like that (anything but 0.4 exactly)

Goury, 2018-05-08:
@fieldOfView this is terrible workaround, it will affect accuracy negatively

@yuyat correct workaround is to set nozzle width explicitly in configuration file, like this:
```
[values]
machine_nozzle_size = 0.4
```

maverickccie, 2018-05-08:
If I could find the configuration file. It's not in the location Cura help says it should be.

icester, 2018-10-14:
The same in version 3.4 and 3.5 set to any but up on reload sets it to 0.5 mm nozzle...

Theoristos, 2018-12-02:
Still present in 3.4.1 
Can be fixed manually by config editing: https://github.com/Ultimaker/Cura/issues/2933#issuecomment-443516254
Looks like **Ghostkeeper** solution report https://github.com/Ultimaker/Cura/issues/3338#issuecomment-375624635 is not fully implemented  in 3.4.1 yet.