Ultimaker/Cura #3339: Cannot use X-Axis grab-arrow to move object. Y and Z work fine.

Careless, 2018-02-17:
I installed CURA 3.2.1. yesterday, and have had an issue with being unable to move an imported model by using the red "move model" arrow.  I can use the blue and green arrows to move objects in the other directions, and when the X axis arrow does work, it puts the model some 10,000 to 25,000mm off the build plate. Attempting to move it back onto the build plate or to zoom out and grab the red arrow to do so will cause CURA to immediately show the MacOS spinning beach-ball mouse cursor and eventually crashes with no response, requiring a force quit.

Seems to happen more often with more than one model on the build platform. I am attempting to multiply and arrange a model so that they nest into one another so that I can print many copies.

"**_ENSURE MODELS ARE KEPT APART_**" is turned off.

**Application Version**
3.2.1

**Platform**
MacOS High Sierra 10.13.3

Vandrasc, 2018-02-19:
Hi @Careless, did you first use the camera view (right bottom corner icons) before trying to move the model around? Or did you zoom in on the models? I am trying to figure out how to reproduce your issue. Thank you!

Careless, 2018-02-19:
are you referring to the **top-right** buttons that change the viewport to various angles? 

my answer to that would be "maybe". it happened a few times during different instances, so I can't recall if I had pressed the camera view buttons.

I tried it again this morning and it doesn't seem to be happening right now- but the movement of items using the arrows is very choppy. it will go from say 14mm X to 50mm X and jump around while I try to move it. Even if I drag it slowly it's still quite jumpy.

It's not a computer performance problem. I didn't have this issue in CURA 3.1... or 3.2.0 now that I think about it. Nothing on my computer has changed. I haven't installed any new applications that would take up additional GPU/CPU/RAM resources either.

ChrisTerBeke, 2018-03-02:
Could you try to reproduce the issue on video? We also can have a look at the debug logs if you upload them here.

Ghostkeeper, 2018-03-26:
Maybe a log file would help as well.

ChrisTerBeke, 2018-05-23:
No new info was added, cannot reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke