Ultimaker/Cura #3340: Feature request: slow down print speed x% at overhang angle x°

ozfunghi, 2018-02-17:
Currently printing an object that needs a slow speed for a sharp overhang where i don't wish to use supports. If printed slowly enough, supports are not needed. However, this means the entire model needs to be printed slowly.

It would be beneficial for such occasions if - similarly to supports at x° overhang - the print speed could be reduced at x% of normal print speed for x° overhangs. That way the entire model can be printed at an acceptable speed, except for the sharp overhang, where the print will slow down.

smartavionics, 2018-02-17:
I am currently working on adding settings (speed/flow/fan) that are used when printing bridges (100% overhang). One of the settings (currently called Bridge Wall Max Air Gap) controls whether smaller amounts of overhang (less than 100%) will use the bridge settings. This sounds very much like what you are asking for.

ozfunghi, 2018-02-17:
I don't have much problems with actual bridges (between two points of a model) 100% flat/horizontal. But when at one end of an overhang, there's nothing to support it, you can easily get PLA curling. This can be helped by better cooling hence slower print speed. So if your setting allows me to set the speed for say 75° overhangs, i'm happy, because that's exactly what i wanted. In such a case you can get away without supports, get a cleaner print, use less plastic, and even in spite of needing to slow down for those parts, get a faster print overall since you don't need the supports in those places.

smartavionics, 2018-02-17:
Do have a sample model with an overhang that should be printed slower that I could try? If so, what speed do you print the overhang at?

smartavionics, 2018-02-17:
I realise that I already have a suitable model that you provided for issue #3335. I am printing that using the above mentioned settings and it's not looking too bad. I think the part cooling on my printer could be better.

smartavionics, 2018-02-18:
So here's a screenshot showing how the overhang areas of that model can be printed slower using the (as yet unavailable) bridging code:

[image]

ozfunghi, 2018-02-18:
That's be great, since those are the parts that can curl when printed too fast. A picture of two 40cm prints, one was printed at 45mms inner walls and 30mms outer wall, and the other at 25mms inner walls and 15mms outer wall. The difference is huge, but also the difference in print time.  So this feature could make me print this part even slower, while keeping the rest of the print at 50mms or higher.
[image]

smartavionics, 2018-02-18:
I think it should work well for that model once the settings are tuned. I hope that the bridging code will make it into the next release but sometimes it can take longer depending on whether any of it has to be changed/improved and how busy the Cura developers are.

ozfunghi, 2018-02-18:
Sure man, i'm happy it's being considered at all.  Thanks for your effort so far!

nokemono42, 2018-03-06:
I think this feature request is more important then just trying to print impressive bridges.

There are many print models that can be printed without supports if only the printer knew to slow down at overhangs. Minimum layer time isn't enough and its effect is almost non-existent. The Phil A Ment Mascot is a perfect example: https://www.thingiverse.com/thing:2557603. No matter the scale or minimum layer time the hands curl when printing with PLA. 

[image]

While proper cooling with PLA is necessary, slicing an object that only the heavily modded printers can handle seems ridiculous. Overhang detection already exists and initial layer print speed is a commonly adjusted setting. An "Overhang Speed," be it mm/s or percentage of print speed, would dramatically improve print quality for entry-level printers.

ozfunghi, 2018-03-06:
That's exactly why i made the request, nokemono42. See my picture above. But if you have to set the overall print speed at the "lowest common denominator" (meaning, speed for good overhang results), it will take forever to print a large object. It may not have been exactly what Smartavionics had in mind when he started his bridging code, but from what he tells us, it's basically the same thing in practice. This could really be a big deal, as it could eliminate supports for even very steep overhangs while improving quality AND reducing print time (by eliminating supports). Better quality, less wasted filament, faster print time. What's not to like :-)

Ghostkeeper, 2018-03-09:
That bridging code is merged now so I'd like to hear whether that was sufficient to solve this overhang problem once the 3.3 beta is out.

ozfunghi, 2018-03-09:
> That bridging code is merged now so I'd like to hear whether that was sufficient to solve this overhang problem once the 3.3 beta is out.

Well, basically any print with an overhang as the two above will tell you that. I'll try to remember to post here first time i try it out. I really think this could be a big deal, like i stated above. Thanks for the effort!

smartavionics, 2018-03-09:
Hi @ozfunghi, when 3.3 beta comes along and you get to try this new feature you need to set 2 bridge settings to make it print the overhangs slowly.

1 - the minimum bridge length needs to be really short (I would just set to zero).

2 - the bridge wall max overhang setting can be altered to control how much overhang is required to
trigger the slowdown. Reducing that from the default of 100% will make it slowdown on less overhang, have a play with it, you will soon get the idea.

Please let us know if it is useful.

Ghostkeeper, 2018-03-29:
The beta is released (see software.ultimaker.com/Cura_open_beta) so maybe I can bump this up a little bit to have you guys experiment with it?

ozfunghi, 2018-04-05:
First try... it's doing absolutely nothing :-(

I probably should have tried smartavionics propositions, but for me, it should be a lot simpler, it's not clear at all. I was hoping for something as clear as the support settings: At "x" overhang, change "y1" print speed & "y2" fan speed

smartavionics, 2018-04-05:
I recommend that you use the feedrate display in the layer view to see where the print speed has been reduced in the overhang regions. Make sure you are doing the things mentioned in https://github.com/Ultimaker/Cura/issues/3340#issuecomment-371830354

ozfunghi, 2018-04-05:
Yup, the feedrate preview is helpful... i had never used/seen it before. Well, i'll try it out a next time i guess.

VanessaE, 2018-05-19:
While it's good that Cura has comprehensive bridging settings, it's not appropriate to use them for overhangs.  One can stretch a PLA bridge across a wide gap at say 30-60 mm/s (depending on the specific filament), but you would probably never want to use those kinds of speeds on a steep overhang.

AbeFM, 2018-06-15:
The machinery is there, but it should be broken out/duplicated specially aimed at the problem suggested.

smartavionics, 2018-06-15:
As it stands at the moment, you could have one profile for bridging and another for overhangs. I agree that it would be difficult at the moment to print both bridges and overhangs on the same model optimally.

AbeFM, 2018-06-15:
...which illustrates why it would be beneficial to have two sets of
settings to use those tools correctly in each circumstance.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

smartavionics, 2019-01-03:
> "bridge wall max overhang" - I can't find this in cura 3.6

It's been replaced with the new overhang settings:

[image]

stemaDev, 2019-01-04:
so this is now separated from the bridging setting and I cannot set the fan to be turned on while doing an overhang?

smartavionics, 2019-01-04:
Correct, all you can do is change the print speed for the overhang.
The good news is that it will no longer start/end a wall in an overhung region.

smartavionics, 2019-01-04:
It could be made to change the fan speed but I am not currently contributing any new features to Cura.

stemaDev, 2019-01-04:
I'm printing abs and petg, where I don't want any fan activity, only when bridging / printing a fast layer/ having an overhang. Too bad, thanks anyway!

ozfunghi, 2019-01-04:
> 
> 
> I'm printing abs and petg, where I don't want any fan activity, only when bridging / printing a fast layer/ having an overhang. Too bad, thanks anyway!

For bridging, you can enable the fan in the bridging sections. 
[image]

For overhangs, maybe slowing down dramatically, could help without cooling.

stemaDev, 2019-01-05:
Thanks. For bridges I know I can, and I do, I just mentioned the cases I would need stronger cooling for materials where cooling is not or just barely used. And the problem with slowing down below a certain speed, that the nozzle's distance is so low that I the result would be the exact opposite of what I'm trying to achieve.

AbeFM, 2019-04-05:
I've been having issues with small ~spherical islands on top of support (specifically PLA islands on  PVA supports in my case) curling at the edges and knocking over my support towers.

Would these settings effect that, could I use this to get rid of curling as the OP demonstrated so well with the Benders?

Beherith, 2019-08-10:
Dont mean to necro a ticket, but what the poster desires is implemented and works perfectly in cura 3.4.1+
The most important aspect is to set minimum bridge wall length to 0.01mm, as high detail models will have wall segments quite a bit shorter than the default 5mm. 
Here are my settings:
[image]

And the results displayed in the feedrate layer view:
[image]

Beherith, 2019-08-10:
Sorry for double post, but @nokemono42  and @ozfunghi , this is exactly what you desire, no? 
[image]

ozfunghi, 2019-09-09:
> 
> 
> Sorry for double post, but @nokemono42 and @ozfunghi , this is exactly what you desire, no?
> [image]

I now simply put the overhang  wall angle & speed to something like this
[image]
... and it works perfectly like i intended. I don't dabble in the bridge settings because they feel too conveluted.

I do have one question. Why are overhangs only identified when going outward, and not when going inward? For closing the top of a model (i print large hollow objects most of the time, without supports or infill), this would work wonders for the roof of such models. Is this something that could be achieved with the bridge settings? Now i have to set a high top layer count, and things still get messy when those start to curl. Currently testing a slow top layer speed to close the top of a 420mm print... which now will take 4 days to finish.

[image]

Ghostkeeper, 2019-09-09:
> Why are overhangs only identified when going outward, and not when going inward?

It would be annoying to have it slow down and do weird things on the inside if there is infill to support it anyway.

ozfunghi, 2019-09-11:
> 
> 
> > Why are overhangs only identified when going outward, and not when going inward?
> 
> It would be annoying to have it slow down and do weird things on the inside if there is infill to support it anyway.

Like i said, i (and i'm sure i'm not the only one) print hollow without any infill as much as possible. So, having the option would certainly be very welcome.

vjvarada, 2019-11-14:
Since this related a recent issue I raised #6656 , asking here as well.

When printing with soluble supports, materials like ABS and Nylon like to curl up when being deposited on top of the soluble supports, as they default to normal printing speeds

printing slower print speeds, similar to how the first layer is laid down on the print bed, may fix the problem.

I've played around with the overhang wall and bridge settings, but it seems that when supports are added, the overhangs are not recognized anymore, and these settings don't apply.

I turned bridging off, and just using the overhang wall speed settings am getting the following behavior when supports are off ( as expected)

[image]

When I turn supports back on, then it ignores the overhang wall settings.
[image]

Is there any way to preserve the behavior of wall overhang settings or bridging settings for supported skin layers as well?

smartavionics, 2019-11-14:
My Cura builds now have an extra bridge setting which tells it to ignore support so that should do what you want. My Cura builds are available at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. You can install my releases alongside the UM releases with no conflict. Please read the README.md file at the above location.

vjvarada, 2019-11-14:
Brilliant @smartavionics ! Im excited to test it out, and shall post results.

vjvarada, 2019-11-14:
@smartavionics Great results with ePVA+ and ABS with the bridging settings above supports. 
In the part below, the nozzle used to corner the two prongs on each side of the part fast and the feature would curl as a result. I had to set the print speed at 8mm/s to get it to print using the bridging setting to get it to work reliably. 
[image]

In your settings, I could only enable "Bridge Over Support" with the "Enable Bridge Settings" being active, but could not use the "overhang wall speed" setting without briding being active for overhangs above support. 

In my test, I turned the bridge, flow, density and coasting settings to values that better-represented bottom skin, then an actual bridge, which IMO the "Overhang Wall Speed" should take care off.

smartavionics, 2019-11-14:
> In your settings, I could only enable "Bridge Over Support" with the "Enable Bridge Settings" being active, but could not use the "overhang wall speed" setting without briding being active for overhangs above support.

Ah yes, I will take a look at that. I guess I need to add a separate "overhangs ignore support option"?

vjvarada, 2019-11-14:
Imo, "Overhangs ignore support" for overhang wall speed option makes more sense than it being ignored during bridging. 

by definition, bridging shouldn't need support, and should use settings that best for printing between two pillars/features where support isn't required.  This would probably be complete horizontal overhangs, such as over holes on the face of the part, small features supported on two sides, etc.

Overhang Wall Speed, would play a part when printing parts where bridging doesn't apply, in overhangs that are steep or cannot be bridged because it would dangle in air, or like in my case, horizontal features printed over support material too large to be bridged.

I'm not sure if there is any way to make a distinction between the two in software.

Anyway, for now, I have what I need to get the job done, I'm pretty happy to be able to print ABS with water-soluble supports (HIPS was a real pain to remove), thanks a lot for your help!

smartavionics, 2019-11-14:
> by definition, bridging shouldn't need support, and should use settings that best for printing between two pillars/features where support isn't required.

I thought that originally but bridging over support was requested by a user and so now it's provided.

ozfunghi, 2019-11-14:
This features is (imho) one of the best features in Cura, because it has enabled me to print most of my prints without having to use supports and get a much nicer finish to boot. I use it all the time, and it is turned on by default in all my profiles. I'm extremely happy that Smartavionics picked up on my initial request.

However, in order to take "the next evolutionary step", i want to ask, if it is possible to "add" new parameters to the feature? In order to get more "steps". By which i mean, add more than one rule. An example:

1* Overhang wall angle: 40°
1* Overhang wall speed: 80%

2* Overhang wall angle: 55°
2* Overhang wall speed: 50%

3* Overhang wall angle: 65°
3* Overhang wall speed: 25%

smartavionics, 2019-11-15:
FYI, my next Cura build will provide separate settings for controlling whether support is ignored below bridges and overhangs.

vjvarada, 2019-11-15:
Much appreciated @smartavionics .  I'd test using support blockers, get some of the overhangs to print with bridging settings, and the rest with the overhang wall speed over support settings. 

Although down the road it would be nice to have automatic detection of bridgable overhangs, working with support blockers can be annoying.

Ghostkeeper, 2019-12-03:
Since the original issue here is implemented, I think we can close this issue. Thanks Smartavionics for your continued support!

closed 2019-12-03 by Ghostkeeper