Ultimaker/Cura #3343: Cura 3.2.1: badly formatted "printing time"

cmock, 2018-02-17:
**Application Version**

Cura 3.2.1 Appimage

**Platform**

Debian 9 / amd64

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Print something and watch under "Monitor", and see negative minutes:
[image]

I've also seen the "minus" sign in the "seconds" field.

**Actual Results**

Negative minutes shown.

**Expected results**

positive minutes.

**Additional Information**

GMagician, 2018-02-20:
I confirm issue, I had same yesterday

Ghostkeeper, 2018-03-12:
I'm unable to reproduce this. When putting any value as input in our `getPrettyTime()` function that's used there, the minutes and seconds are always positive.

When I tried actually printing (via USB) the printing time also stayed positive.

cmock, 2018-03-12:
``getPrettyTime()`` was the right hint: look at the code in the 3.2 branch and at commit af89209cdeb31eec4abf9ebcb30266239ac35676 which probably has fixed this issue...

closed 2018-04-11 by Ghostkeeper