Ultimaker/Cura #3345: Can't set two extruders

mark-hahn, 2018-02-18:
**Application Version**
3.2.1

**Platform**
Ubuntu 16.0.4 unity

**Steps to Reproduce**
set up machine settings for flashforge creator dual

**Actual Results**
The drop-down for number of extruders only has one extruder with no other option

Here are my machine settings.  It shows one extruder and I cannot select two.
[image]

Kinglyboiz, 2018-02-18:
which printer are you using? I am using Tevo Tarantula with dual extruder and have the same issue. I am able to select the customer FDM printer and chose a second extruder, but I'm creating another issue where I cannot print with the second extruder. Cura seems to lock up. 

Running 3.2.1 on MAC OS

ChrisTerBeke, 2018-02-19:
That's because the printer definition for the Tevo Tarantual has no extruders defined, meaning it will fall back the default: 1 FDM extruder. They (or the person who's maintaining the Cura profiles for Tevo) should update the definition (or you can make a PR to fix it). We usually don't maintain 3rd party machines as we can't test with them.

Details: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/tevo_tarantula.def.json.

Ghostkeeper, 2018-03-12:
To prevent confusion, I've now hidden this drop-down for printers where no more than 1 extruder is defined: https://github.com/Ultimaker/Cura/commit/aae8a31f00cf9db2b6190a43dbf1658db64311cf

closed 2018-03-12 by ChrisTerBeke

Ghostkeeper, 2018-03-12:
We've been assuming that if you are technical enough to modify your printer to add additional extruders that it normally doesn't have, you'd also be technical enough to modify the JSON files of your printer definition in Cura's installation.

That's not always the case of course. But you could use Custom FDM Printer as an alternative.

Festivejelly, 2018-06-02:
This isnt very user friendly. Also assuming that someone knows their way around a printer should also know their way around JSON and how the formatting works is a bit ignorant tbh. I feel you should be able to add this through the UI.

paalex, 2020-05-15:
For those that decide to tweak Cura on their own, you can find the printer configuration file here (macOS):
```/Contents/Resources/resources/definitions/tevo_tarantula.def.json``` inside ```Ultimaker Cura``` app package

Replace ```tevo_tarantula.def.json``` with your printer's config filename.

Edit this part to add additional extruders:

```"metadata":
    {
        ...
        {
            "0": "tevo_tarantula_extruder_0",
            "1": "tevo_tarantula_extruder_1"
        }
    }```

paalex, 2020-05-16:
@TheAssassin @FN5969