Ultimaker/Cura #3347: Strange Signs and Spelling in Setting-Fieldnames

xiaoandy80, 2018-02-18:
**Application Version**
3.2.1

**Platform**
Mac

**Qt**

**PyQt**

**Display Driver**
NVIDEA Newest Mac update

**Steps to Reproduce**
just starting cura and drop down settings.

**Actual Results**
Hollow Out Ob'ects
'nfill

**Expected results**
Hollow Out Objects
Infill

**Additional Information**

fieldOfView, 2018-02-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/3277, fixed in https://github.com/Ultimaker/Cura/pull/3320

ianpaschal, 2018-02-20:
Thanks @fieldOfView! @xiaoandy80 In the issue fieldOfView referenced, you'll find manual download links for a patch, which will be included in the next official build (use the manual download for now so Cura is usable 😉).

closed 2018-02-20 by ianpaschal