Ultimaker/Cura #3348: Heating done but no print

Kinglyboiz, 2018-02-18:
I am having an issue with the latest Cura 3.2 and 3.2.1 releases on MAC OS.

I have Tevo Tarantula with dual extruder. 

First issue is that the Tevo Tarantula printer settings does not allow the second extruder to be selected. How can i enable the second extruder on the TEVO Tarantula? 

I have managed to configure the customer FDM printer with my TEVO settings, using a second extruder, but when i go to print, it seems Cura will lock up. I have attached the log file from cura which shows an error "Could not parse hotend temperatures from response"  @ 2018-02-18 11:02:25 
my heatbed and hot ends heat up to the set temp 60/60 and 205/205. the LCD shows heating done, but CURA doesnt show the second extruder temp. The CURA GUI doesnt update the heatbed temp as soon as it reaches the set temp and initiates the hotend temp. my printer and CURA App can sit there all day and nothing is done to print. 

[cura.log](https://github.com/Ultimaker/Cura/files/1734869/cura.log)

**Application Version**
3.2 and 3.2.1
**Platform**
MAX OS 

**Display Driver**
Intel HD Graphics 3000
  Type:	GPU

**Steps to Reproduce**
set customer printer with 2 extruders. diameter on each extruder set to 1.75 and nozzle 0.4. Printing with EXTRUDER #2 does not work. There is no temp displayed and print will not proceed. 

**Actual Results**
Cura does not display updated heat bed temp. Cura does not display EXTRUDER #2 temp. CURA Does not initiate the print after heating is done, as per LCD.
**Expected results**
After LCD shows heating done, i would expect the print should start

Kinglyboiz, 2018-02-18:
sorry am new to here was not aware i should have just followed the format that was provided.

ChrisTerBeke, 2018-03-02:
Could you try printing the same G-code with another host software (like Repetier Host)? Then we can start excluding things.

ChrisTerBeke, 2018-05-23:
No new info, cannot reproduce, 3rd party machine. Closing.

closed 2018-05-23 by ChrisTerBeke