Ultimaker/Cura #3351: Feature request: Accelereation after retraction prime

vvranjek, 2018-02-18:
I print a lot of precise single-walled objects with relatively high speeds. The only problem I have is under extrusion after a move with a retract. Extra prime distance is not an ideal solution for me at all and slowing down print speeds increases printing time immensely. 

I believe that accelerating slowly after priming on retraction will completely fix the issue.

I appreciate  your good work and hope to see this feature in the future.

BagelOrb, 2018-03-06:
Introducing such accelerations might solve that problem, but influence totally different parts of the printing process, which makes it difficult to get control over the process and ensure a given print quality.

Ghostkeeper, 2018-03-09:
This sort of compensation gets into the realm of predictive modelling the flow of material, which can be done with some sort of compensations here and there but in general is a bit beyond the scope of what a slicer can achieve right now...

Mulakulu, 2024-08-26:
I had the exact same idea as you [vvranjek](https://github.com/vvranjek), just 6 years later. I was considering adding this to a python script that look for retractions and slow down the acceleration for a certain amount of distance after the extrusion, but was hoping there was some setting already implemented. Seems like there isn't I guess.