Ultimaker/Cura #3353: Wanhao i3+ Print won't start issue. Cura 3.2/3.2.1

rs-mn, 2018-02-18:
**Application Version**
3.2.1

**Platform**
 Windows 10 (Dell Studio i7 Laptop. 8gb ram.) Wanhao Duplicator i3 Plus.

**Qt**
Downloaded from Website.

**PyQt**
 Downloaded from Website.

**Display Driver**
 ATI Mobility Radeon HD 4650

**Steps to Reproduce**
 Load part. Slice. Select Print to USB. Then the platform heats, but extruder doesn't after platform reaches temp. Doesn't show temps in the Cura monitor display.  Print doesn't start. Also, Preheat selection doesn't work or if it does, it doesn't show the temps on the monitor tab. Extruder and bed temps always read zero in the monitor tab. BTW, Cura 3.1.0 runs perfect. This issue was present in Cura 3.2.0 and 3.2.1.

**Actual Results**
Bed heats up. Print won't start. Shows printing on monitor tab, but it never is really printing. 

**Expected results**
 Bed and nozzle should heat up and print should start. Also temp/time info should be accurately seen in the Cura Monitor tab.

**Additional Information**
[image]
[image]

fieldOfView, 2018-02-19:
Please upload your cura.log somewhere we can download it and inspect it. Go to Help -> Show configuration folder, and cura.log is in the folder that opens.

diegopradogesto, 2018-02-20:
In the log I see this:
`2018-02-18 15:15:15,201 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]: Traceback (most recent call last):
2018-02-18 15:15:15,201 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 178, in findContainers
2018-02-18 15:15:15,217 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]:   File "C:\Program Files\Ultimaker Cura 3.2\plugins\LocalContainerProvider\LocalContainerProvider.py", line 60, in loadContainer
2018-02-18 15:15:15,217 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]:     container.deserialize(f.read())
2018-02-18 15:15:15,217 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerStack.py", line 363, in deserialize
2018-02-18 15:15:15,217 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerStack.py", line 377, in deserialize
2018-02-18 15:15:15,217 - ERROR - [(10200)-MainThread] UM.Logger.logException [86]: Exception: When trying to deserialize i3 Plus, we received an unknown ID (wanhao_i3plus) for container`

Probably you got the **wanhao_i3plus** printer definition from somewhere and in some point this file was deleted. Now Cura can't find the file and it is using the basic **fdmprinter** definition. 

The wanhao_i3plus is not a definition we support yet, so we can't do anything for you in this case. What you can do is try to get the definition file (just like the first time you created the printer) and start from scratch.

I hope this helps.

closed 2018-02-20 by diegopradogesto

rs-mn, 2018-02-21:
I'm curious how 3.1 can be uninstalled and reinstalled and work fine with this printer, but 3.2 won't work when installed.

Ghostkeeper, 2018-03-09:
3.1 didn't have the definition either.

We don't upgrade definition files automatically when installing a new version of Cura. Maybe you had a definition file but because we change the format for these files continuously that definition is now no longer in the correct format for Cura to read it.