Ultimaker/Cura #3357: List of Gcode macros/variables

samr28, 2018-02-20:
Is there a list of gcode macros/variables somewhere? I couldn't find anything in the wiki, google searches, or searching through the code. I need one for {IF_COOLING} and {COOLING_FAN_SPEED}. I would think that those already exist but I'm not sure what they are called as there is no documentation on this.

**Application Version**
3.2.1

**Platform**
Windows 10

fieldOfView, 2018-02-20:
There are no conditionals (IF_*). The variablenames can be found in resources/definitions/fdmprinter.def.json

samr28, 2018-02-20:
Yeah, it seems like this gcode is just out of date or something. Those lines don't seem to do anything... I still can't get the cooling fans to run. Cura should add a `M106 S###` to the top of the gcode to turn it on but it seems to be absent even when enable cooling is checked. Should I make a separate issue for this?

```
;FLAVOR:Marlin
;TIME:3658
;Filament used: 4.9183m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.2.1
T0
M190 S60
M104 S210
M109 S210
M82 ;absolute extrusion mode
```

samr28, 2018-02-20:
wait I'm dumb. It does that on layer 1 >_>

closed 2018-02-20 by samr28