Ultimaker/Cura #3358: [Cura 3.2.1] Sub-menus inside drop-down menus don't work

markhoe, 2018-02-20:
**Application Version**
Cura 3.2.1

**Platform**
Windows 8.1

**Steps to Reproduce**
Use a sub-menu inside one of the drop down menus (e.g., settings > material> ... )

**Actual Results**
Nothing can be selected and the blue highlighting shown when hovering over something, doesn't appear

**Additional Information**
[Example of this happening here](https://www.youtube.com/watch?v=y81V_boWOs0&feature=youtu.be)

fieldOfView, 2018-02-20:
As a workaround, use the cursorkeys to navigate the menu.

ianpaschal, 2018-02-21:
We'll look into it. On macOS it's working, so we'll need to figure out what OS specific conditions are causing this.

Ghostkeeper, 2018-03-12:
I've seen this reported before (but can't find the issue any more).

It only occurs on Windows. And if I recall, the cursor keys don't work either.