Ultimaker/Cura #3361: [3.2.1] No temp shown for 2nd nozzle

GMagician, 2018-02-20:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10

**Display Driver**
ATI unknown ver

**Steps to Reproduce**
Create a custom printer with 2 nozzle

**Actual Results**
No temperature on 2nd nozzle, just on 1st

**Expected results**
Temperature value shown for both nozzles

As noted above with a printer with 2 nozzle everything can be managed but temp of 2nd nozzle is not shown

fieldOfView, 2018-02-20:
Needs more info:
What printer, what printer definition (Custom FDM Printer?), are you using USB printing or OctoPrint, what do the logs say?

GMagician, 2018-02-20:
Yes, started from Custom FDM printer and using USB, I'll check log

GMagician, 2018-02-20:
> 2018-02-20 21:51:14,880 - WARNING - [(1236)-Thread-10] USBPrinting.USBPrinterOutputDevice._listen [567]: Could not parse hotend temperatures from response: b'ok T:21.47 /0.00 B:24.00 /0.00 T0:21.47 /0.00 T1:21.16 /0.00 @:0 B@:0 @0:0 @1:0\n'

I think that is is almost clear...
I don't know what should be correct message from Marlin to be correctly parsed.
If you can help me I could post a PR to it

fieldOfView, 2018-02-20:
I currently don’t think the message from the firmware is wrong, but the parsing makes a wrong assumption about something. I screwed up a little hiding the actual error. I hope to have time thursday to try to reproduce the issue and create a fix for it.

GMagician, 2018-03-28:
Duplicated report. I close mine since #3563 is newer and has been labeled as bug

closed 2018-03-28 by GMagician

kirilledelman, 2018-03-29:
@GMagician when you say "everything can be managed but temp of 2nd nozzle is not shown", does that mean you were able to preheat, and start print, despite this bug? I'm 99% done putting together a dual extruder upgrade, and encountered this bug while testing. Wasn't sure if I'll be able to print.

GMagician, 2018-03-29:
@kirilledelman sorry but I'm also starting up my printer. Just 1 extruder is working now.
I made some gcode test and I saw that T0/1 commands are generated but I saw some strange things with temperature. It lower extruder temp before changing to new extruder but temp is not the same with T1 and T0. Not investigated further

kirilledelman, 2018-03-30:
@GMagician I'll finish my setup, try printing with two nozzles, and report back.