Ultimaker/Cura #3362: issue while slicing several objects on the same plate

notmine642, 2018-02-20:
Hello, I have added 2 items on the plate and it top slicing at the top of the lowest object

[image]

diegopradogesto, 2018-02-20:
I think this is a duplicate of #3318

ianpaschal, 2018-02-20:
Yup. @diegopradogesto is correct. Devs, see CURA-4967.

closed 2018-02-20 by ianpaschal

notmine642, 2018-02-20:
yes correct, duplicate !

*Pascal Hierckens*
Group Engineering and Construction
Industrial Data Analytics
T: +32 2 264 2131 - M: +32 475 73 63 53

[image: SOLVAY]
Rue de ransbeek 310
B-1120 Bruxelles

*www.solvay.com <http://www.solvay.com/>*

Connect with Solvay:

<https://twitter.com/SolvayGroup> <http://www.linkedin.com/company/solvay>
<https://www.facebook.com/solvaygroup>
<http://www.youtube.com/user/SolvayGroup>

-- 

*This message and any attachments is intended for the named addressee(s) 
only and may contain information that is privileged and/or confidential. If 
you receive this message in error, please delete it and immediately notify 
the sender. Any copying, dissemination or disclosure, either whole or 
partial, by a person who is not the named addressee is prohibited. We use 
virus scanning software but disclaim any liability for viruses or other 
devices which remain in this message or any 
attachments.*******************************Ce message, ainsi que toute 
piece jointe, est exclusivement adresse au(x) destinataire(s) nomme(s) et 
peut contenir des informations confidentielles. Si vous recevez ce message 
par erreur, merci de le detruire et d'en avertir immediatement l'emetteur. 
Toute copie, transmission ou divulgation, integrale ou partielle, par une 
personne qui n'est pas nommee comme destinataire est interdite. Nous 
utilisons un logiciel anti-virus mais nous denions toute responsabilite au 
cas ou des virus, ou tout autre procede, seraient contenus dans ce message 
ou toute piece jointe.*