Ultimaker/Cura #3363: [3.2.1][Feature request] Change view angle using as origin the mouse position

GMagician, 2018-02-20:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10

This is a feature request (if not already present but I've not found option).

When using mouse to change view angle you get more intuitive results if object is rotated using mouse position as origin. It seems that some other point is used as origin of rotation (maybe plate origin)?

ianpaschal, 2018-02-20:
It's true that there's typically two ways to do rotation, one being around the camera position (as though you're turning your head to look around) and one being around some target (as though you're moving around the target watching it).

The problem is that the mouse is not a point in 3d space (it is in fact, a line). You would need to specify some point along that line (the 3rd coordinate) to specify an origin of rotation. Where would that be? The first intersect with an object? What if your mouse position is in empty space? By default, Cura puts this imaginary "target" point at the center of the build plate, and moves it as you pan around.

I'm all for "more intuitive" but what you're experiencing is one of the fundamental issues of trying to navigate 3D space with a 2D interface. The current solution is also, in my opinion, a thousand times easier to use in CAD applications than "first person controls" which rotate around the camera origin like in a video game.

Side project idea: AR/VR Cura.

So: I'm going to close this for now because rotating around the mouse position, is simply not mathematically possible. The next closest thing is the current behavior. But! If what you were referring to is an option to have to rotate around the camera origin (not the mouse), I can reopen it as a potential feature to toggle between. But I do not think it's very intuitive or would be very in-demand.

closed 2018-02-20 by ianpaschal

fieldOfView, 2018-02-20:
> Where would that be? The first intersect with an object? 

Yes.

> What if your mouse position is in empty space?

Rotate around the camera origin.

The whole camera system could do with an overhaul, making zoom an actual zoom (instead of a dolly), or a proper dolly (instead of a dolly that is limited by the rotation origin).

ianpaschal, 2018-02-20:
> Yes.
> Rotate around the camera origin.

Switching between first person and third person views depending on what your mouse is hovering?

Yuuuuuuck.

I agree about the zoom but I think the rotation system is exactly how it should be.

fieldOfView, 2018-02-20:
I would like to see a prototype before making a "yuck" judgement. An alternative to rotating around the camera origin could be to fall back to rotating around the center of the buildvolume (or the center of the models on the build plate).

GMagician, 2018-02-20:
@ianpaschal good point about 2D/3D coordinate systems, I missed just one little thing :-D
I see that using different 3d programs (I can name them if required) has a more intuitive results than the one seen here, not sure how they achieve this.
What I see when I try to rotate a piece is that such piece goes always out of screen and is not what I naturally expect from it.

ianpaschal, 2018-02-20:
@fieldOfView "Yuck" is based on like 12 years of CAD, and 15 years of gaming. Seriously. It's not a good way to do it. I've seen so many attempts at different approaches in dozens of different software and it's really not a practical way to do it. Like I said, making a toggle between "orbit" and "fly" controls is maybe valid so people can pick their preference but trying to jump between the two is _definitely_ "yuck." Also suppose you try to rotate by clicking and doing a bunch of short drags (not much mouse space, small trackpad, etc.). The rotation point then will jump around between each of these movements.

This is one of these cases where I know I can be annoying to others because I feel like I've perfectly prototyped it in my head and others can't see it failing inside my head, but yeah. I'm willing to bet cash it's not a good way to do it. Maybe @Appesteijn can comment on if this sort of thing is possible to devote time to building a prototype.

@GMagician: Would be helpful to know which methods you do find intuitive.

fieldOfView, 2018-02-20:
Bro-tip: don’t try to out-experience someone with more grey hair than you ;-)

ianpaschal, 2018-02-20:
I'm not trying to out-experience you, Aldo. It's nothing personal. I'm just explaining why I have such a strong feeling against the idea; it's not laziness, or fear of feature creep, etc. It's just going against the sum of all my personal experience over the years.

GMagician, 2018-02-20:
I think that what I find unnatural is due to where is located the fulcrum of rotation.
Now I have to find out where is located in soft I usually use and then understand the differences with cura...

GMagician, 2018-02-23:
Ok I used other programs (OnShape just to let you know) and it seems that it changes fulcrum of rotation.
When you zoom an object to see details and you try to rotate detailed area with cura is almost impossible to do because of fixed fulcrum.
On shape seems to adjust fulcrum on visible area