Ultimaker/Cura #3365: [Bug] Lost usable build area

compatibilizer, 2018-02-20:
There is a serious logic error in the way that Cura limits build plate area by the amount of helpers (skirt/raft/support/brim etc.). Let me explain;

Slicer expects all of the part (bounding box) to be within the build volume, and that is correct, nozzle can travel only within this volume.

But, shrinking the build plate by the amount of raft/brim etc. is wrong. Parts can be slanted, and base of a part might be laying more inwards. Let me show with pictures;
[image]
My part is using 15mm raft, so Cura decreases my usable XY plane by that amount, this is the most leftward I can position my part in Cura
[image]
But my part is slanted and my base is 10mm inwards, so my raft protrudes the bounding box only by 5mm instead of full 15mm that Cura assumes it will. 

In this example, we have effectively lost 10 mm of usable space because of the margin. Hope you can fix this serious error soon...

Regards,

ianpaschal, 2018-02-20:
I don't think that's what's happening.

The grey area on the build plate is area which the extruder can't reach. If you have 2 extruders next to each other, you'll notice if you switch between left and right, the "usable area" shifts left and right.

compatibilizer, 2018-02-20:
Nope, I have one extruder on this machine. Furthermore, when I disable raft in this example, the gray area totally disappear.

ianpaschal, 2018-02-20:
Can you share some more details in this case? Is it a custom FDM machine? For me, a custom FDM machine creates these margins based on the print head size settings and behaves as expected.

compatibilizer, 2018-02-20:
When I took the screenshots, machine was a Prusa i3 on Cura v2.7, I have posted the on Ultimaker forum back then.

Now I use a custom FDM machine on Cura v3.2 and it is still the same. It has got nothing to do with the machine definition, gantry size, etc. this is just the way Cura handles the brim/raft/skirt/support etc. 

The amount of helpers (brim,raft,...) is equal to the amount that Cura will restrict your build area. But your part can be slanted, in which case, those helpers will not totally protrude from your bounding box, but only from your base.

Here, try it yourself with this simple slanted geometry on Thingiverse and do it like the screenshots I posted above. **Put some raft that will protrude further than the slant** and realize you can not position that edge as far as your printer physically can.
https://www.thingiverse.com/thing:626033

diegopradogesto, 2018-02-20:
Cura calculates the gray areas using the bounding box, but the raft is created with a margin based on the base, not on the bounding box. Cura shows the worst case scenario for that.

I don't think there is a easy way to fix it, because it depends on the models that you load in the build plate.

diegopradogesto, 2018-02-20:
For this exact model that you show in the picture, I suggest to rotate it 180 degrees and then it works as you expected.

ianpaschal, 2018-02-20:
@compatibilizer Yup, you're right. Sorry I misunderstood. Off the top of my head I'm not sure what the right solution would be. Ideally this would be computed by the "footprint" instead of the bounding box, however if a model is floating or bridging, this would become very difficult to predict/compute. As @diegopradogesto says, it depends on the model and while this is not ideal for that model, doing it another way would be really non-ideal in other cases.

compatibilizer, 2018-02-20:
@diegopradogesto I am trying to fit the maximum number of parts on the build plate possible, so it is not a solution, then I will face the same issue on the other end.

compatibilizer, 2018-02-20:
@ianpaschal yeah I think you're right, I am not familiar with Cura engine code, but it should not be a difficult task to calculate footprint, rather than mindlessly limiting the build area. I know for a fact that Makerbot Desktop (I believe it is now called Makerware?) was able to do this 5 years ago

ianpaschal, 2018-02-20:
As we always say, comparing one software to another is never really a useful comparison. It says nothing about the underlying mechanics of those softwares.

Seeing as using only the footprint instead of the convex hull (that's the bounding box you're referring to; it's not really a box) would introduce problems for plenty of other models I see two possible solutions:

1. Compute 3D hulls (similar to how video games create hulls around objects for physics calculations). This would be the "smart" solution which would allow you to really densely pack your build plate. Basically it would allow for a distinction between "am I off the build plate yet" at the base vs. at the tip of the overhang in your model. The downside? Lots of work to implement.

2. Add a power user setting to turn off the margins and rely on the user to make sure they're not trying to print off the build plate. This one is easier to implement (I think) but would get a low priority as it's not that useful to most users.

ianpaschal, 2018-02-20:
[image]

The 3rd monkey head is an example of a "convex hull" as would be used in physics engines, and could in theory also be used to check intersections in Cura.

compatibilizer, 2018-02-20:
Sorry didn't mean to offend or compare anything, just wanted to point to an example. 

Regarding the "convex hull" approach, I believe that is not the simplest solution to the problem. Doesn't the engine first create a modifier mesh (then assign different slicing properties) to the helpers like rafts and supports?

Why not use those modifier meshes combined with the part mesh itself, then try to fit that combined mesh into the build volume. If you can somehow extract the modifier mesh geometry, this should be the simplest solution.

ianpaschal, 2018-02-20:
> Sorry didn't mean to offend or compare anything, just wanted to point to an example.

It's ok. It just comes up a lot so it kind of gets a canned answer.

> Why not use those modifier meshes combined with the part mesh itself, then try to fit that combined mesh into the build volume. If you can somehow extract the modifier mesh geometry, this should be the simplest solution.

This would require that you have to first send to CuraEngine, slice (and generate all the other features), _then_ check if it fits in the build volume, and if not, repeat the process. With complicated models that require lots of slicing, this would be a mess. Also imagine you've loaded up the build plate (as you are intending) and then send to CuraEngine, doesn't fit, then try to move the model over a bit, runs into another model, try again, etc. etc. etc.

It's just a lot cleaner to try and preemptively avoid that situation altogether by creating those margins.

Since what you're experiencing is the expected behavior for now (even if not ideal for your use case) I'm editing this to be an enhancement.

Ideally, yes, 3D intersection testing should be done with 3D hulls, instead of 2D as it's done now. But I'm not sure when we would get around to it.

GMagician, 2018-02-20:
OT
@ianpaschal in my configuration I have limits X=196 Y=199 but gray area is almost 10mm x side (hence white area is almost 176x179)
If on 2nd extruder I add a 13mm offset gray part decrease on X max side by 13mm but don't care about which extruder is active, but it should

Maybe I have wrong configuration somewhere, not surfe if I missed something

ianpaschal, 2018-02-20:
@GMagician If you load a model and then set it to extruder one or two you should notice the grey area shifting. Selecting the extruder at the top of the settings panel does not do this as that is merely toggling which extruder's material/core/other settings are currently being edited. If you are not seeing that behavior, please open a separate ticket for it.

diegopradogesto, 2018-02-20:
The gray areas are also for the brim, raft and skirt. If you have "None" selected as your preferred adhesion method, then you will see no gray areas. That's what we were talking about.

GMagician, 2018-02-20:
@diegopradogesto it's brim...I think I have to learn more and more things about this program. Only 13mm hotend offset will stay there even with 1st hotend selected (that has offset 0)

compatibilizer, 2018-02-21:
Just for the sake of staying on the topic, call it whatever but this is not a feature request, the behavior is already there for the right reasons, but I don't agree that this is expected behavior. I would not, under any circumstance, expect the slicer to disable my usable build area where I could be printing on.

I don't know how many times this happened to me before, it was just that I realized it later. Inch-by-inch population of the build plate is probably common practice for many professionals and academics, who use the machine for prototyping or research, and according to [Statista](https://www.statista.com/statistics/739182/worldwide-survey-3d-printing-applications/) these are the most common use cases for a 3d printer.

Regarding how to implement the "smart" solution, I still think convex hull approach will make life harder than necessary. We just need to divide the conditional check into two pieces. There are helpers that follow the "body contour" (ooze shield, draft shield, support,..) and there are helpers that follow the "touching base" (raft, brim, skirt,..), do the conditional checks separately for both groups and allow positioning when both hold true. I think it should be easier to implement?

ianpaschal, 2018-02-21:
> I don't agree that this is expected behavior. I would not, under any circumstance, expect the slicer to disable my usable build area where I could be printing on.

Ok, but we do. Expected behavior just means the code is doing what it's supposed to. It's supposed to limit the build area to preemptively avoid gcode generating off-plate.

> There are helpers that follow the "body contour" (ooze shield, draft shield, support,..) and there are helpers that follow the "touching base" (raft, brim, skirt,..), do the conditional checks [...]

No, because positioning is done _before_ those structures are generated. That's why if you nudge your model after slicing, it has to reslice. You're essentially proposing a guess and check (and adjust) solution which is not clean code and probably not that easy to implement. It's also only a half measure to a problem which has a "proper" solution.

I'd actually really like to implement the convex hulls though as it's pretty standard in video games and there's plenty of literature out there on how to do it efficiently.

compatibilizer, 2018-02-21:
> ...because positioning is done before those structures are generated. That's why if you nudge your model after slicing, it has to reslice. You're essentially proposing a guess and check (and adjust) solution 

I believe you misunderstood my suggestion this time. The suggestion I made yesterday requires slice, check and repeat  as you explained, and I see that it would be inefficient.

This time though, my suggestion is a pre-slice check, just like it does already. Just make 2 seperate groups, think about it like having 2 different gray area. First gray area is "contour based offsets" and if the bounding box fit inside that is a pass. Second gray area is "footprint based offsets" and if the touching base fit inside that, then it is a pass. If both conditions pass, allow that position.

ianpaschal, 2018-02-21:
Ah ok. So;

**Current:**
- The bounding box must be within the build volume minus brim width.

**Proposed:**
- The bounding box must be within the build volume, and
- The footprint must be within the build volume minus the brim.

Sound right?

It's maybe less difficult to implement than the convex hulls but also not an easy fix. We'll put it into consideration.

ianpaschal, 2018-02-21:
This is actually a duplicate of [#1453](https://github.com/Ultimaker/Cura/issues/1453). I'll close that one and we can continue the discussion here.

compatibilizer, 2018-02-21:
Sounds about right, I have detailed further below. There may be others also but as far as I am aware, these are the helper features that introduce the mentioned offset to the build plate;

**Current:**

- The bounding box must be within the build volume minus _**brim/skirt/raft + support X/Y distance + draft shield/ooze shield**_ width.

**Proposed:**

- The bounding box must be within the build volume minus **_support X/Y distance + draft shield/ooze shield_**
- The footprint must be within the build volume minus the **_brim/skirt/raft_**

GMagician, 2018-02-21:
Sorry to bother with semi OT, if I have to open a new issue please let me know.
My problems are also related to working permitted area (gray borders) and some are related to brim.
Two nozzles and 13mm X offset on 2nd. If brim is activated when you change active extruder 13mm gray area stays wrongly on xmax. If I disable brim 13mm gray area move from xmin to xmax depending on selected nozzle.
These behaviors are wrong. 
1) changing active extruder has to change permitted area even when brim is active
2) changing active extruder not always need to change permitted area.
My printers home X and Y at negatives positions, X0Y0 is bed origin and dimension in printer configurations is maximum X and Y reachable by 1st extruder. Selecting extruder 2 should correctly limit Xmax but selecting 1st extruder shouldn't limit xmin (X home at -16mm and 0 is always reachable)

ianpaschal, 2018-02-21:
@GMagician Yes, as I said, if the above comment hadn't explained/fixed it, please open a new topic. You can just copy and paste what's here and I'll remove the comments from this discussion.

compatibilizer, 2018-03-12:
Then can we at least have the option to disable the enforcement and allow manual positioning? It could come with a warning message "Use at your own risk"

Ghostkeeper, 2018-04-06:
Cura's philosophy is that we know for certain that printing it could cause permanent physical damage to the printer, then it won't allow it. Similarly we've forbidden the use of PLA through the BB core of the Ultimaker 3 because we know that it will clog pretty quickly.