Ultimaker/Cura #3367: [3.2.1] Can't create new material from scratch

GMagician, 2018-02-20:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10

**Display Driver**
ATI unknown ver

**Steps to Reproduce**
Preferences -> Configure Cura... -> Materials -> Create

**Actual Results**
Nothing happens

**Expected results**
Some dialog/windows where to insert new material data

**Additional info**
All other keys works as expected (Import, Export, Duplicate)

ianpaschal, 2018-02-21:
Interesting. This was working for me yesterday while fixing some bugs. I don't think it's related, but curious nonetheless.

Can you post a screen shot of what you're seeing? Should help to diagnose if it's a rendering issue or a data issue.

GMagician, 2018-02-21:
This has been done with linus mint but same results in windows 10
[screenrecorder.mp4.zip](https://github.com/Ultimaker/Cura/files/1743628/screenrecorder.mp4.zip)

ianpaschal, 2018-02-21:
Ok. First guess is it's a UI problem as there's no empty boxes or labels appearing (which would likely be the case if the UI was fine but there was some missing data/model).

Can you share a log from when this happens? With that we can diagnose with more certainty.

GMagician, 2018-02-21:
Don't know where to find log in linux... Cura is an AppImage and \var\log doesn't contain anything about cura. I'll check window version this evening

noisemaker00, 2018-02-21:
I have same issue, then searching in the 'materials' folder in
`%APPDATA%/Roaming/cura/<version>/materials`
I found the new material has 2.85mm as `material diameter` and due to I set a printer with 1.75mm filament support, I suppose the incompatible materials will be hidden.

The material diameter was set automatically by Cura, I didn't see any field to set it, so when I pressed 'Create' material has been created and hidden immediately due to filament diameter incompatibility.
Result = it seems 'Create' button not working.

GMagician, 2018-02-21:
I think problem is similar to @noisemaker00. In my case no material is created but in log I have:

> 2018-02-21 22:40:20,617 - DEBUG - [(3936)-MainThread] cura.Settings.ContainerManager.createMaterial [851]: Unable to create a new material by cloning Generic PLA, because it cannot be found for the material diameter for this machine.

X-PRESSIVE, 2018-03-08:
Same problem here. Cura 3.2.1, Windows 64 bit:

In the materials window:

- Can't change any existing material (diameter is 2.85mm everywhere but I'd need to change it to 1.75mm)

- Cant create a new material (nothing happens)

- Can duplicate a material, but as soon as I change a setting, it suddenly disappears

- Can't import an existing material from previous Cura version. It says "Import successfully", but the imported material is not shown.

tamepenguin, 2018-03-11:
Same here. Logs say 

> 2018-03-11 08:54:55,062 - WARNING - [(41684)-MainThread] UM.Qt.QtApplication.__onQmlWarning [267]: file:///C:/Program Files/Ultimaker Cura 3.2/resources/qml/Preferences/MaterialView.qml:410: Error: Cannot assign to non-existent property "properties/diameter"

(Windows 10)

rtxg, 2018-03-17:
Try this:
Go to settings, printer, manage printers.
Click "machine settings", then click the extruder tab.
Check whether"Compatible material diam" is set to 2.85mm. If it is, change it to 1.75mm and then go back to manage materials and see if that fixes things.

I had the exact same symptoms as listed by X-PRESSIVE and after hunting through, and experimenting with all the options, changing the above setting fixed it.

Ghostkeeper, 2018-04-16:
This has been fixed in 3.3.

On the master branch it has been broken again, but the error is completely different so it's a different issue. The error I'm getting on master seems to be an infinite recursion caused by lazy loading that was implemented more recently.

closed 2018-04-16 by Ghostkeeper

hackra76, 2018-04-19:
As this was closed and I had the same exact problem before, I today found out, that for some bizarre reason, the default filament diameter in Settings - Printer - Machine settings was setup to "175" not "1.75"...
After I changed the value to the correct 1.75, all the materials populated the selection table ad I was able to import and create new material.
Hope this helps someone with the same problem.

0xc0dab1e, 2020-06-06:
I have run into similar problem of no material could be created. The workaround has been:

1) Set the diameter at 2.85 mm in the machine settings
2) Import some `.xml.fdm_material` profile from Internets
3) After step (1), the material appears in the Materials list. 
4) Edit material according to manufacturer's specifications. When diameter is edited to 1,75 - it disappears from the materials list.
5) Go back to printer settings and modify supported filament diameter to actual 1.75. Then the material with corresponding diameter should appear in the materials list. 

An example of some `.xml.fdm_material` template for PLA:
```
<?xml version="1.0" encoding="UTF-8"?>
<fdmmaterial xmlns="http://www.ultimaker.com/material">
    <metadata>
        <name>
            <brand>Generic</brand>
            <material>PLA and PLA</material>
            <color>Generic</color>
        </name>
        <GUID>506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9</GUID>
        <version>1</version>
        <color_code>#ffc924</color_code>
    </metadata>
    <properties>
        <density>1.24</density>
        <diameter>2.85</diameter>
    </properties>
    <settings>
        <setting key="print temperature">200</setting>
        <setting key="heated bed temperature">60</setting>
        <setting key="standby temperature">175</setting>
        <setting key="soften temperature">175</setting>
        <setting key="probe temperature">175</setting>
        <setting key="wipe temperature">175</setting>
        <setting key="part removal temperature">75</setting>
    </settings>
</fdmmaterial>
```