Ultimaker/Cura #3369: Materials

Thisismydigitalself, 2018-02-20:
Why when i change materiel i am asked to chose whether to keep or discard settings which seems to not directly related to the properties of the material?

[image]

And.. while I'm at it, trying to make some order in my pool of materials i noticed it is not a simple task. can't delete materials, can't always create or duplicate others. it seems not to work that well.

ianpaschal, 2018-02-21:
This is because when you switch to another material, the default settings for that material/quality profile are loaded which includes not only properties of the material but also how the material is used.

So this is the expected behavior. I'm closing this because it shouldn't actually block any of the workflow and we likely will not change it for the foreseeable future.

closed 2018-02-21 by ianpaschal

Thisismydigitalself, 2018-02-21:
>  the default settings for that material/quality profile are loaded which includes not only properties of the material but also how the material is used.

This makes no sense to me. just say'n.

Ghostkeeper, 2018-03-12:
So which settings would you say are directly related to the material? Print temperature, surely. And build plate temperature, I suppose. Overhang angle, I guess, because some materials sag more than others. And related, the support Z distance, because if a material sags more then you need a higher Z distance to be able to break the support off. And of course then also the support density to still be able to get nice overhangs if the material sags a lot. Similarly, you'd need to modify the infill density or the skin will sag just as much on the inside of the model as on the outside.

And some materials extrude better than others. Can you really achieve that 0.3mm layer height on a 0.4mm nozzle using PVA? Probably not. Better reduce the layer height for PVA then, so layer height is also really a material setting.

Of course if a material extrudes better then you'd want to use that to get faster print speeds, so the print speed is a material setting too. And if you're going to have faster print speeds then the infill will be printing faster too. But with faster infill you're going to see that infill shine through the walls as well, so we'd better make the wall thickness a material setting too.

And then you see that all 3 settings in your screenshot are actually material settings.

There is no clearly defined border between settings that are specific to a material and settings that are not. Most settings will ideally change based on your material.

Thisismydigitalself, 2018-03-12:
I understand. I was simply asking if it makes sense to include and add to the list of my preferred butter's ingredients list and nutrition facts the recommended amount and spread thickness as well so i will know how to apply it on my breakfast toast. I guess when it comes down to terminology there is no right and/or wrong.  

Case closed.