Ultimaker/Cura #3370: Missing Sidebar and all controls

techmastermike, 2018-02-20:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10

Having trouble with the interface with Cura 3.2.1 - the sidebar is missing! It should be on the right, and changing the view settings only increases the blank box size. Pi is configured and shows webcam, but printer is not accessable!
[image]
[image]

fieldOfView, 2018-02-20:
Installed any plugins lately?

Can you go to Help -> Show configuration folder and upload the file “cura.log” somewhere so we can have a look for error messages?

techmastermike, 2018-02-20:
Only plugin was Octoprint, which I had to browse for to enable, but it kept the API, so the webcam works.

Here's the log - thank you!
[cura.log](https://github.com/Ultimaker/Cura/files/1742037/cura.log)

fieldOfView, 2018-02-20:
This is the problem:
```
2018-02-20 12:55:57,470 - WARNING - [(10604)-MainThread] UM.Qt.QtApplication.__onQmlWarning [267]: file:///C:/Program Files/Ultimaker Cura 3.2/resources/qml/Sidebar.qml:125:5: Type Label unavailable
2018-02-20 12:55:57,472 - WARNING - [(10604)-MainThread] UM.Qt.QtApplication.__onQmlWarning [267]: file:///C:/Program Files/Ultimaker Cura 3.2/qml/QtQuick/Controls.2/Label.qml:40:1: plugin cannot be loaded for module "QtQuick.Templates": Cannot load library C:\Program Files\Ultimaker Cura 3.2\qml\QtQuick\Templates.2\qtquicktemplates2plugin.dll: Access is denied.
```

I don’t know why that would happen though. Do you have an antivirus application on your system?

techmastermike, 2018-02-21:
Antivirus only for email. I reinstalled 3.1 and it seems to be working fine. Hopefully other users won't have the hiccup!

fieldOfView, 2018-02-21:
I think it is better to keep this issue open, or it will surely still be an issue in Cura 3.3.

edrose, 2018-02-21:
I'm having the exact same issue on Arch Linux in version 3.2.1 which only updated a few days ago. The issue should be kept open since downgrading is not a fix.

fieldOfView, 2018-02-21:
@Edrose, could you also check your logs?

edrose, 2018-02-21:
Yes of course - here you go. 

The line that struck me was this one 
`2018-02-21 15:30:20,581 - WARNING - [(140552838796608)-MainThread] UM.Qt.QtApplication.__onQmlWarning [267]: file:///usr/share/cura/resources/qml/Sidebar.qml:5:1: module "QtQuick.Controls" version 2.1 is not installed`

[cura.log](https://github.com/Ultimaker/Cura/files/1744569/cura.log)

fieldOfView, 2018-02-21:
Are you using the AppImage? If not, you should be able to install QtQuick Controls with your package manager.

edrose, 2018-02-21:
It looks like a packaging issue for me. The arch package depends on `qt5-quickcontrols` but if I install `qt5-quickcontrols2` it works.

edrose, 2018-02-21:
There is already an arch bug report for my issue here https://bugs.archlinux.org/task/57559?project=5&string=cura

hroncok, 2018-03-21:
[Requiring qt5-qtquickcontrols2 fixed this in Fedora as well.](https://src.fedoraproject.org/rpms/cura/c/de5259a14ea2990f4241ec7b55a0391189f45416?branch=master)

closed 2019-01-07 by nallath