Ultimaker/Cura #3371: [3.2.1] Wrong computation on printable area

GMagician, 2018-02-21:
**Application Version**
Cura 3.2.1

**Platform**
Virtualboxed Linux Mint hosted on Windows 10

**Steps to Reproduce**
Setup a custom fdm printer with 2 extruders and dimension X=196 Y=199 and offset X=13 on 2nd extruder, brim width is default to 8mm
Activate brim and activate extruder 1 and 2, deactivate brim and do same extruder activation

**Actual Results**
When brim is activate then gray area is always xmin=8 xmax=8+13 even when extruder 1 is selected.
When brim is not active gray area xmin/xmax has 8mm and 21mm swapping depending on active extruder

**Expected results**
Printable area limits for brim and extruder offset should work independently.
Another expected things is homing position configuration. I have X-16 at home (X0 is bed origin) so when I select extruder 1 there is no limit due to extruder offset.
Cura has only printer dimensions (and I set extruder 1 limits) it miss homing positions.

ianpaschal, 2018-02-21:
Relates to #3365

BagelOrb, 2018-03-06:
Ultimaker will not invest time in solving issues for 3rd party printers.

There might be community contributers like @fieldOfView out there who are willing to help on this one, but to keep things tidy we will close this issue.

closed 2018-03-06 by BagelOrb

GMagician, 2018-03-06:
@BagelOrb ok about support to custom printers but part of the issue is indipendent. When you change nozzle extruder offeset is always from one side when brim is active and it swap when no brim..
If Ultimaker has multi extruder printers you need to fix it

BagelOrb, 2018-03-06:
It also swaps when you swap with which extruder you print the brim.

GMagician, 2018-03-06:
In my test if you have brim activated gray area doesn't swap, only when none is selected gray area swap.

BagelOrb, 2018-03-08:
Yeah but did you also test switching the extruder with which you print the brim?

GMagician, 2018-03-08:
you're right! There is no issue at all