Ultimaker/Cura #3372: [Extrusion Multiplier] Add this parameter in the 'Material' panel, due to it is strictly related to the filament brand/material/etc. you are using

noisemaker00, 2018-02-21:
Please add the option to set an `extrusion multiplier` (like the one in Slic3r) in the materials panel.
This parameter is very related to the type of material you are using, and may be differences from one brand to another. 
It cannot be a global variable to set. Will be very cool if when I switch material then this is changed too.

I made some test with ABS/PLA/PETG/TPU both with Cura and Slic3r and this parameter may really improve prints.

nallath, 2018-02-21:
You mean flow? Cura already has this.

pfjason, 2018-02-21:
I believe the issue is that Flow is attached to the print profile and not the material profile. Moving the setting would be helpful.

ChrisRiddell, 2018-02-28:
Under Material on the side panel their is Flow (not sure if hidden by default might have to use the search box) it's attached to the quality profiles.

eg: Slic3r a extrusion multiplier of 1.1 would be 110% flow in Cura.

fieldOfView, 2018-02-28:
In the Cura architecture, a setting cannot be "moved". All "levels" of profiles can in theory contain all settings (*). If a setting has a value in a higher "level", it overrides the value set in a lower level. This is not going to change.

*: materials currently support only a subset, but this is only a limitation of the xml format. Here's a PR that adds support for adding arbitrary settings to the material level: https://github.com/Ultimaker/Cura/pull/2333

ChrisRiddell, 2018-02-28:
It's already in the 'Material' category on the side panel called flow. (eg Multiplier of 1.1 would be 110% flow)

The Flow setting is attached to the quality profiles not the materials profile so you would need to set the flow for each quality.

superjamie, 2018-03-28:
> The Flow setting is attached to the quality profiles not the materials profile so you would need to set the flow for each quality.

This is the problem. Flow rate should not be a property of quality Profile, it is a property of the Material. One should be able to print with the same quality Profile using different Materials and have the flow rate adjusted properly per Material.

The inheritance described above is fine, that's not going to change and doesn't really have to.

Now that @fieldOfView has #2333 in master, it just needs someone to add the Flow parameter to the Materials UI. The Material Flow setting would then override a default Profile Flow setting, and this Issue will be resolved.

Ghostkeeper, 2018-04-24:
> Flow rate should not be a property of quality Profile, it is a property of the Material.

Ah, but if you use a 0.06mm layer height you'll want to have this setting tuned completely different from what you have it when you use a 0.4mm layer height, because at 0.4mm your extruder won't be able to keep up with your preferred 130% flow rate. This sort of argument can be made for 50% of the settings we have out there.

Maybe one day it'll be possible to put any setting in the material profiles. Right now that is possible when editing the files but not in the interface.

Nandox7, 2019-03-04:
Revisiting this as I moved to cura and was looking for a way to compensate the differences in filament width.
I was actually using the filament width in the material details (that I ended up realizing that has no use) and later found Flow option. Main idea is to compensate the differences in filament width, I have some going as low as 1.70 and does make a difference if you don't increase the flow.

So I agree with @superjamie this should as well a property of the material. Independent of the layer height/quality if you filament is thinner than expected you'll need to compensate for that.

batesman1, 2019-04-22:
+1 for this want

fieldOfView, 2019-08-25:
See the Material Settings plugin in the Marketplace.