Ultimaker/Cura #3373: Crash when slicing in cura 3.2.1 with invalid extruder temperatures

ftsabrov, 2018-02-21:
**Application Version**
 3.2.1

**Platform**
 Windows 10

**Display Driver**
Doesn't matter.

**Steps to Reproduce**
 Using a dual extrusion printer, activate a profile where Extruder 0 is off (0C) and Extruder 1 is set to some value. Attempt to slice a model that is set to print with E0.

**Actual Results**
Popup saying "CuraEngine has stopped working" appears. Windows prompts user to close the software, Cura itself stays open and is still able to slice other models correctly.

**Expected results**
This should have a popup that warns the user that the model is attempting to be sliced with an inactive extruder assigned.

**Additional Information**
Using a non-Ultimaker printer, so no material profiles are used.

Vandrasc, 2018-02-21:
I think it may be related to https://github.com/Ultimaker/Cura/issues/3309, where setting the Temperature to 0 degrees will cause a crash. As a workaround we advise setting the temperature to 0.1. 
If indeed related than it is already fixed and available in the next release.

BagelOrb, 2018-03-06:
Please attach your .curaproject.3mf file (and rename it to .curaproject.3mf.zip)

BagelOrb, 2018-03-06:
Cannot reproduce with an UM3 or with a Builder medium dual extrusion printer.

project file is required to reproduce.

ianpaschal, 2018-05-28:
Can't reproduce, needs more info, never got it.

closed 2018-05-28 by ianpaschal