Ultimaker/Cura #3375: Bad default setting for 0.8AA print core (rafts)

pswiss, 2018-02-21:
**Application Version**
 3.2.1

**Platform**
Ultimaker 3

When I use the AA0.8 print core, the Build Plate Adhesion -> Raft -> Raft air gap is set to 0mm, which causes the parts to be very attached to the raft (the gap is 0.3mm when using a 0.4AA print core)

Vandrasc, 2018-02-22:
Hi @pswiss, which material are you using? For the most materials brim is the default adhesion and also recommended to print with brim.

pswiss, 2018-02-22:
I am using PLA. You are correct in saying that brim is the default, but when I switch to raft, certain raft settings are automatically set. 

My reason for using a raft is that I've had bad luck printing relatively large hope with a brim. Because brim only prints on the outside, I find that the inner portion tends to peel up

-Petras

From: Andreea Scorojitu

Sent: Thursday, February 22, 2:01 AM

Subject: Re: [Ultimaker/Cura] Bad default setting for 0.8AA print core (rafts) (#3375)

To: Ultimaker/Cura

Cc: Petras Swissler, Mention

Hi @pswiss, which material are you using? For the most materials brim is the default adhesion and also recommended to print with brim.

Vandrasc, 2018-02-22:
Oh, did you tried to disable Brim Only on Outside setting? It will print some brim lines on the inside of the model as well (for hallow models). 
For most of the models, is it ideal to print with 0mm air gap when printing with a 0.8 core, of course, this is model dependent. I will run some prints, in the meantime, as workaround, could you change your airgap value? 
Thank you!

pswiss, 2018-02-22:
Thanks for the help, I hadn’t seen that setting

 

Thanks,

 

-Petras

closed 2018-02-26 by ianpaschal