Ultimaker/Cura #3376: BUG. Cannot place Prime Tower inside the model

karabas, 2018-02-22:
**Application Version**
CURA 3.2

**Steps to Reproduce**
Model a two color cylinder or thor or vase without bottom. 
Try to set primetower coordinates inside the model. 

**Actual Results**
Warning model does not fit build volume 
[image]

**Expected results**
Normal slicing

**Additional Information**
I turned off "ensure  models are kept apart" but it does not help

alekseisasin, 2018-02-23:
@karabas Could you please save the model as project and upload to Github, then we can check?
From this project we will extract your printer data, model and used settings.

ianpaschal, 2018-02-23:
This is in fact not a bug but expected behavior. For each mesh, in order to prevent collisions, Cura computes something called a "convex hull" which is basically the tightest shape around the mesh which is made of only convex angles. It's used a lot in graphics, games, physics engines, etc. Here's an image from wikipedia showing a convex hull around some arbitrary points:

https://upload.wikimedia.org/wikipedia/commons/b/bc/ConvexHull.png

This means that your prime tower cannot be inside the volume of this hull. Normally this is a pretty simple limitation which makes the code far, far simpler and less prone to errors, but I can see how with your round build plate, that's not an option.

Luckily, @Ghostkeeper proposed a workaround for you:

You can manually model a prime tower as part of your mesh, and then set the LayerStartX and LayerStartY settings to be _inside_ the prime tower area. In this way each layer of your print will start by printing your manual prime tower.

Hope this helps.

closed 2018-02-23 by ianpaschal

karabas, 2018-02-26:
Why "ensure models are kept apart" option does not any effect? But works with any model mesh I need to overlap,

ianpaschal, 2018-02-26:
Sorry, I'm afraid I don't really understand your question. Keeping models apart option is intended to keep models from touching, for for where you place the prime tower.

karabas, 2018-03-01:
with turning off this option you can intersect meshes f.e. for generation custom support etc. Therefore it is possible! That is why I thought it must help im my case.

Ghostkeeper, 2018-03-23:
The question is, if "Ensure Models Are Kept Apart" is disabled, why does it still disallow printing something that intersects with the prime tower?

The reason is that Cura right now doesn't make any distinction between obstructions on the build plate from the printer, such as clips that hold the bed down or the switching bay, and printed obstructions such as the prime tower or initial prime position. Colliding with solid pieces in the geometry of the printer can cause damage to the printer. Colliding with printed pieces such as the prime tower just causes overextrusion. Overextrusion is worth a warning state in Cura but shouldn't prevent slicing.

To model a prime tower you can load two cubes and scale them to the height of your print. Set one cube to be printed with the second extruder. Then merge the two cubes (Ctrl+Alt+G). Cura will automatically alternate the layers between the two extruders. Then you only need to make sure that these cubes are printed at the beginning of a layer by setting the Layer Start X/Y positions onto the cube.

karabas, 2018-03-27:
Ok, thanx I will try such a tower