Ultimaker/Cura #3378: Cura 3.2.1 nozzle width is resetted to 0.5 every time start it

Goury, 2018-02-22:
**Application Version**
3.2.1

**Platform**
Ubuntu 16.04 using appimage
Windows 7×64 using installer from website

**Display Driver**
Nvidia 364 for Ubuntu
Nvidia 388 for Windows

**Steps to Reproduce**
Start Cura.
Even if I set nozzle to 0.4 in printer settings (and it works just fine for the session), as soon as I quit the application and start it again nozzle is set back to 0.5.

**Actual Results**
Nozzle is set to 0.5.
So if I forget to check it I will get ugly blob of plastic instead of print.

**Expected results**
Once set to 0.4 should remain 0.4.
As it was in 3.1.

**Additional Information**
I set profile to 0.4 with Cura 3.1 and it worked just fine with 3.1.
But 3.2.1 shows 0.5 and makes it back to 0.5 every time I start it.

fieldOfView, 2018-02-23:
Have you used the 3.1 beta or only the release versions? The beta contained a bug that put the configuration in a conflicting state.

Thisismydigitalself, 2018-02-23:
https://github.com/Ultimaker/Cura/issues/3338

Goury, 2018-02-23:
@fieldOfView one that was in releases here on github

Thisismydigitalself, 2018-02-24:
Meanwhile, Is there a workaround for fix'n this?

Thisismydigitalself, 2018-02-26:
https://github.com/Ultimaker/Cura/issues/3378#issuecomment-368252092

fieldOfView, 2018-02-26:
I still don't see a straight-up answer if you used the 3.1 beta. The releases page has a tag for the 3.1 beta that would screw up your configuration. In the words of Jerry Maguire: "Help me help you".

Goury, 2018-02-26:
@fieldOfView No, it wasn't beta.
And you could tell us how to detect such screw and what to with it, because other people can find this info useful.
So please point us to related manual or other form of information.

fieldOfView, 2018-02-26:
Does the same also happen when you add a new printer, and change the nozzle size on that?

Goury, 2018-02-26:
Yes, it does. Yet instead of 0.5 it's 0.35 every time.

Thisismydigitalself, 2018-02-26:
Cura 3.2.1 - Added new printer and this one keeps the value(s) i set, including nozzle size.
For what it worth my current printer which does keeps going back to 0.5 each time i launch it (Custom FDM) was created the same way on a clean install.

Thisismydigitalself, 2018-02-27:
Here's one for you:
Just tried changing my "keep reverting back to 0.5 nozzle" printer's nozzle to 0.2, 0.3, 0.6 and guess what - they were all saved and open as saved.. but when i set it to 0.4 - it reverts back to 0.5.

Beats me.

Zwaubel, 2018-02-28:
Having the same issue with 0.4 mm nozzle size. Would be very nice to have that fixed!

Thisismydigitalself, 2018-02-28:
Can you please try my workaround and tell us if 0.399 nozzle size works for you as it works for me?

Goury, 2018-02-28:
.399 workaround kinda works, but I don't like how it looks, I want it to be 0.4
So please label this as bug already and fix it some day.

closed 2018-03-02 by ChrisTerBeke

SuraTech, 2018-03-03:
guys I still could not solve the problem .... I still have the size of the nozzle 0.35 ... but I use the printer "KOSSEL PRO" ... if you select "СUSTOM" will also be 0.5

Zwaubel, 2018-03-04:
Sorry for the delayed response, I'm somehow not getting the notifications in time... 
The workaround does "solve" the issue. But as Gourd mentioned, it looks bad.

Ghostkeeper, 2018-03-23:
This was marked as a duplicate of #3338. Please continue conversation there.

diegopradogesto, 2018-06-28:
We are keeping track of this issue in #3773. It should be now fixed in master branch. Waiting for testers to confirm it fully works.