Ultimaker/Cura #3380: Visibility of execution time not readable in german version

Diavolothecat, 2018-02-23:
**Application Version**
CURA 3.2.1

**Platform**
Windows 10

**Display Driver**
Intel 1920x1080 resolution

**Steps to Reproduce**
Set german language and introduce some SD or USB storage

**Actual Results**
The text "Speichern auf Wechseldatenträger" is too large. So it makes execution time left beside unreadable.

**Expected results**
Proposal: Show Icon instead of text button or use smaller font for large text.

[image]

ianpaschal, 2018-02-26:
Related to #3385.

Ghostkeeper, 2018-03-16:
If you're a native German speaker and you've got a better suggestion for what should be displayed on the button, please let us know.
We hired a translation bureau to make this translation for us but they don't always have the knowledge of how much space they can use.

Diavolothecat, 2018-03-16:
Hello,
I speak that language fluently. I suggest, you replace "Speichern auf Wechseldatenträger" by "Auf Datenträger speichern". By the way, you could also replace "Speichern in Datei" by "In Datei speichern". I understand your difficulties because I'm also working sometimes on UI.

Ghostkeeper, 2018-04-16:
We've now changed it to "Speichern auf Datenträger" which makes it a bit better to read.

closed 2018-04-16 by Ghostkeeper