Ultimaker/Cura #3381: [3.2.1][FR] Temp display always with 2 decimal digits

GMagician, 2018-02-23:
**Application Version**
Cura 3.2.1

When nozzle temp is displayed decimals digits can be 1 or 2 depending on value. The printed text is moving right/left and look not "professional". Maybe forcing 2 digits gives a better look

Ghostkeeper, 2018-03-14:
You mean the display in the print monitor? Or did you mean the temperatures in the material manager? Maybe the ones in the list of settings of custom mode?

The temperature display in the print monitor is rounded to integers, so there are no decimals digits. The total number of digits is going to change though because the integer part gets longer when you get above 10 degrees or 100 degrees.

The temperature in the material manager also doesn't allow decimals as far as I know.

The temperature in the settings list displays as many decimals as needed for its value to be sufficiently accurate. We know though that there is a bug in there that it displays something like "200.0" instead of "200" if the value is derived from a formula. We should fix that...

GMagician, 2018-03-14:
Sorry not to be clear, I'm talking about bed temperature in monitor page

GMagician, 2018-04-13:
I checked 3.3 beta (I don't have anymore 3.2.1) and it seems that decimal on bed is disappeared. Maybe some other fix has changed behaviour

closed 2018-04-13 by GMagician