Ultimaker/Cura #3382: initial printing temperature used for printing

maukcc, 2018-02-23:
V3.2.99

With multi extruder setup at the end of one extruder it starts heating up the other extruder with initial printing temperature, that is good.
When the other extruder starts, it 1st does the extruder_start_code then it gets a temperature command from Cura. That temp command should be normal print temp but is also initial printing temperature. That is not good.

ChrisTerBeke, 2018-03-02:
3.2.99 sounds like a development version. Can you reproduce this using 3.2.1 and a clean user config folder?

maukcc, 2018-03-13:
Yes, this is the latest Cura version and it still persists. 
we see no point in testing this on an older version.

fieldOfView, 2018-03-13:
The point is to see if this is a recent regression or something that has been broken longer. It helps to know where to look when trying to fix it.

maukcc, 2018-03-13:
Tested on 3.2.1 appimage and it is the same there.

Ghostkeeper, 2018-04-11:
I'm not seeing this bug.

I set the initial layer temperature to 201 instead of 200. Then it outputs 201 for the initial layer temperature, not 195 for the initial printing temperature. When making a dual extrusion print it sets the first nozzle to 180 (the final printing temperature) and then stand-by. This is expected. It should be using the initial layer temperature on the first layer.

maukcc, 2018-04-11:
The initial layer is good(it uses the initial layer temp), it is all subsequent layers that use the initial print temp, where it should use the normal print temp.

Ghostkeeper, 2018-05-03:
Still not seeing the bug :(

I took a .gcode file of two cubes printing with different extruders and let loose this regex on it that filters out anything that's not a temperature command, layer change or tool change: `\n(?!(M10[49]|T|;LAYER:)).+(?=\n)` (replace that with an empty string).

When slicing I changed the second extruder to be one degree higher. On the first extruder all temperatures are divisible by 5, so you can recognise it. I added comments in layer 1! The result is this:

    T0

    M109 S205
    ;LAYER:0
    M104 T1 S206
    M104 S185
    T1
    M109 S206
    M104 T0 S100
    M104 S201
    ;LAYER:1
    M104 T0 S190 ;First extruder to initial printing temperature (preheat)
    M104 S186 ;Second extruder to final printing temperature right before the switch.
    T0
    M109 S190 ;First extruder to initial printing temperature (actual heat before printing).
    M104 T1 S100 ;Second extruder to stand-by temperature.
    M104 S200 ;First extruder to normal printing temperature, shortly after printing starts.
    ;LAYER:2
    M104 T1 S191
    M104 S185
    T1
    M109 S191
    M104 T0 S100
    M104 S201
    ;LAYER:3
    M104 T0 S190
    M104 S186
    T0
    M109 S190
    M104 T1 S100
    M104 S200
    ;LAYER:4
    ...

So it appears to be working fine.

ianpaschal, 2018-05-28:
Cannot reproduce, so closing due to it going cold.

closed 2018-05-28 by ianpaschal