Ultimaker/Cura #3383: The main focus for coming releases should be SPEED

hoegge, 2018-02-23:
The current version 3.2.1 is SO slow in all aspects:

1. Layer preview (we need a frame rate above 10 FPS all the time)
2. GUI updates (should happen instantly)
3. Button reaction - should react instantly
4. Loading Cura (>20 seconds) - should take max 5 seconds.

EVERYTHING is so slow it is a pain to use and not worthy for a 5000 USD printer. 

Please either make it work in a speed that is bearable or abandon CURA and adapt Slice3r PE for UM3 instead. It is so counter productive as it works now.

Thanks

ianpaschal, 2018-02-23:
Thanks for your feedback. We are working hard to make these improvements and as you can see from the recent commits there is some _massive_ refactoring going on thanks to @LipuFei and @jackha.

Also, Cura is _free open source software_ so if you feel it's unworthy, you're welcome to contribute. 😉

hoegge, 2018-02-23:
Thanks – looking forward to that. I can’t find the time at the moment – trying to make stuff with my UM3E but would be very fun to participate. 

 

Med venlig hilsen

 

Morten Høgholm Pedersen

 

Mobile: +45 3054 5892
Email:  <mailto:hoegge@gmail.com> hoegge@gmail.com

LinkedIn:  <http://www.linkedin.com/in/mhpedersen> mhpedersen

Skype: m.h.pedersen

ianpaschal, 2018-02-23:
That's fine. We hear your feedback and are aware of it, so I'm going to close this ticket since improving performance is an underlying goal of _every_ release vs. a specific issue which is at some point "done." Thanks for understanding.

closed 2018-02-23 by ianpaschal

hoegge, 2018-02-23:
Thanks. I would say, though, the current performance right now is so slow it is unbearable and something that should be "fixed" for next release and not just gradually improved. Had I knew it was so bad I would never have bought my UM3E where I'm bound to use Cura. The total package is much lower performing than my $500 USD Chinese printer with Slic3r, which is not ok for a $5000 semi-professional machine and this will kill Ultimaker eventually if the "basic combined (and quite expensive) package" of Cura an UM machine does not improve. Hope you see my point, knowing you are not necessarily responsible for that.

jackha, 2018-02-26:
Hi @hoegge ,

Did you use any older Cura? How was the performance? What hardware specs do you have? Do you have example projects / models to share? I suppose they're big models?

It's surprising that the startup is also slow, as 3.2.1 improved the startup time by a lot. If you have a lot of configured printers that you don't use anymore (upgraded from an older version of Cura), then deleting them should also improve startup time.

hoegge, 2018-02-26:
Yes, I think 15 was faster, but things got slower and slower with version 2 and 3. Don't remember, since I used Slic3r and Simplify 3D for quite some time. Big models? I suppose it has more to do with complexity too? Not very big - e.g. 2 parts at 10 x 10 x 5 centimeters. The amount and size of parts doesn't really make a diferrence unless I slice only a 2 x 2 x 2 test cube - then it work quite fine.

PC Specs:
Intel Core i7-3530QM CPU @ 2.4 GHz
32 GB Ram
Only SSD drives
NIVIDIA Quadro K1000M GPU with latest driver (updated within an month)
The system runs Fusion 360 very well.
Slic3r layer preview runs fast and smooth.

I have only two printer: Creality CR-10 and Ultimaker 3 Extended.

Start-up time is the smallest problem if the program worked smoothly when started.

Did a quick comparison to Slic3r PE. Loaded 3 parts each about 10 x 10 x 10 cm on the plate and measured graphics update rate when rotating view:

Orbit and zoom before slicing:
Cura: ~ 20 FPS (Frames per second or update frequency in Hz)
Slic3r: ~60 FPS

Orbit and zoom during slicing:
Cura: 20 FPS
Slic3r: ~20 FPS

Orbit and zoom after slicing in layer preview (and after generating layer preview in Cura)
Cura: 1-2 Hz / FPS = Unbearable
Slic3r: ~ 20 Hz

jackha, 2018-02-27:
@hoegge thanks for the information. It seems that the performance drop is mostly in the layer view? It's surprising that the frame rate in the layer view is so low. I also have a Quadro as well as an Intel GPU on a P50s laptop and I must say that the layer view performance is not great... but still a lot better than the 1-2Hz you have. If I zoom in a lot, on 1920x1080, with Extra Fine quality (0.06 mm layer height), then the frame rate still a few Hz (guess 3-5).

Appesteijn, 2018-03-03:
@hoegge how do you measure the fps?

hoegge, 2018-03-03:
@Appesteijn I use FRAPS [http://www.fraps.com/faq.php](url)