Ultimaker/Cura #3385: [3.2.1] Italian message too long to be read/missing

GMagician, 2018-02-23:
**Application Version**
Cura 3.2.1

**Platform**
Mint hosted on Windows 10

**Steps to Reproduce**
- Load a stl file
- open "scale" dialog, check box messages can't be read because are too long.
- open "Fix model" dialog, messages are not translated

**Actual Results**
Text shown on scale dialog:
'Dimensionamento agg...'
'Dimensionamento unif...'
Text shown on fix model dialog:
Messages and hints in English

**Expected results**
A more readable text, suggested:
'Dim ????' (don't know what 'agg' is)
'Dim uniforme'

ianpaschal, 2018-02-26:
Related to #3380.

macaron16, 2018-02-26:
I suggest to let the choice to change the font size (normal, big, small) / or the possibility to change the size of the split windows (left and right windows)
may be it already in progress ;)

In french translation, we also have many text which are not entire "Optimiser l'ordre ...ression des parois"   for instance in regards of "Optimiser l'ordre d'impression des parois"

With a lower size, I think we could see the entire text with "..."

ianpaschal, 2018-02-27:
> may be it already in progress

We have some other stuff that is taking priority in the next few sprints so it's not in progress (yesterday we created the `Status: In Progress` label for such issues), but it is on our radar and is a topic which is mixed in with other discussions like refactoring how the interface is written when upgrading to QtControls 2.x, or writing a more streamlined theming system.

Ghostkeeper, 2018-03-16:
The current full texts for the two in the original post are:
* "Snap Scaling" -> "Ridimensionamento aggancio (Snap)"
* "Uniform Scaling" -> "Ridimensionamento uniforme"

Can you suggest something shorter?

GMagician, 2018-03-16:
Snap Scaling" -> "Ridimens. aggancio (Snap)" or "Ridim. aggancio (Snap)"
"Uniform Scaling" -> "Ridimens. uniforme" or "Ridim. uniforme"

Ghostkeeper, 2018-03-16:
Thanks. I've made it a bit shorter: https://github.com/Ultimaker/Uranium/commit/1702fecd184622a9151b79d8714d112c6bd9f2ed

There's probably loads of these, but we don't have the time/ability to correct them. We depend a bit on community members to make improvements where they think it's appropriate. If you've seen any more, feel free to post them here. Correcting these takes just a second.

Changing the font size automatically can be considered too but it's not easy to implement and has some issues with the general look and feel. I don't know. For now, let's consider this fixed because the offending text is corrected.

closed 2018-03-16 by Ghostkeeper