Ultimaker/Cura #3386: Cura upgrade error?

kcjoec, 2018-02-23:
Any help on this?
Version: 2.3.1
Platform: Windows-10-10.0.16299
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

fieldOfView, 2018-02-23:
Go to c:\users\[you]\appdata\local\cura, and delete cura.log

Ghostkeeper, 2018-03-14:
In Cura 2.3 it would attempt to load all files in your configuration folder to see if they are profiles that need to be updated to the latest version. One of these files is the log file. If that log file became too big to load in memory, Cura would crash.
We fixed this in Cura 2.4.

closed 2018-03-14 by Ghostkeeper