Ultimaker/Cura #3387: Plugin not adding gcode

black4hawk, 2018-02-23:
**Application Version**
Cura 3.2.1

**Platform**
Windows 10

**Display Driver**
Geforce Experience, newest version

**Steps to Reproduce**
Add Plugin https://www.thingiverse.com/thing:2077884 to file (or any other one)
choose parameters
click close
reslice the file

**Actual Results**
If gcode is checked with a text editor, no "custom" text appears
no changes the plugin is used for get implemented in gcode

**Expected results**
Add additional lines to gcode (depending on choice of plugin)

";LAYER:15
;TYPE:CUSTOM
;added code by post processing
;script: ChangeAtHeight.py
;current z: 3.300000
M83  ;Set extruder to relative mode
G1 E-5.000000 F2400  ;Retract
G1 X0.000000 Y0.000000 Z25.000000 F3000   ;Move head away
M104 S0  ;Turn off extruder heat
M117 Press Continue...
M300  ;Beep
M25  ;Pause
G91  ;Set to relative position mode
G1 X-0.1 Y-0.1 Z-0.1  ; Lock motors
G1 X0.1 Y0.1 Z0.1  ; Lock motors
G90  ;Set back to absolute position mode
M117 Press Continue...
M300  ;Beep
M25  ;Pause
G91  ;Set to relative position mode
G1 X-0.1 Y-0.1 Z-0.1  ; Lock motors
G1 X0.1 Y0.1 Z0.1  ; Lock motors
G90  ;Set back to absolute position mode
M117 Heating extruder...
M109 S210.000000  ;Heat extruder back up for filament change (wait)
M117 Press Continue...
M300  ;Beep
M25  ;Pause
G91  ;Set to relative position mode
G1 X-0.1 Y-0.1 Z-0.1  ; Lock motors
G1 X0.1 Y0.1 Z0.1  ; Lock motors
G90  ;Set back to absolute position mode
;Prime nozzle
G1 E6.000000 F6000
G1 E-5.000000 F6000
M117 Press Continue...
M300  ;Beep
M25  ;Pause
G1 E-5.000000 F2400 ;Retract
G1 X0.553000 Y0.553000 Z3.300000 F3000  ;Move to next layer position
G1 E5.000000 F2400 ;Extrude
M82  ;Set extruder back to absolute mode
G92  E784.434060  ;Set the extrude value to the previous (before last retraction)
M117 Printing...
;CUSTOM Pause Done
"

**Additional Information**
Worked a few times, now it won´t work anymore. Used multiple versions of cura, fresh installs and reinstalled the plugins a few times.

Ghostkeeper, 2018-03-14:
I'm unable to reproduce this. It always seems to add post-processing scripts fine.

We've had problems in the past where some output devices aren't post-processing the g-code before writing it to their device. What output device are you using? Saving to file? To removable drive? Printing via Octoprint? Etc.

black4hawk, 2018-03-14:
Tried saving on pc and SD card, neither worked. Are there any files I can send that help?

Ghostkeeper, 2018-04-12:
Hmm, both saving to file and to removable drive worked fine here :confused:

alekseisasin, 2018-06-29:
Hi @black4hawk , I tested and cannot reproduce your issue. Could you please test again in the latest Cura version.

ChrisTerBeke, 2018-07-30:
Cannot reproduce + long time no answer. Closing.

closed 2018-07-30 by ChrisTerBeke