Ultimaker/Cura #3388: Print via usb does not start

Enano79, 2018-02-23:
I have a Prusa i3  and with this new version when I try to print via usb. Does not do any thing. if I cancel the printing IT got to home. 
Does not start to heating .. nothing. I have try to pre heat and thats works. regads

**Application Version**
3.2.1

**Platform**
Windows 7

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ianpaschal, 2018-02-26:
Hi, can you share your log files or an example project which fails?

Enano79, 2018-03-05:
Sorry I cant find the log file. 
Also I have install again version 3.1.0 and print with that leaving both version installed.
today trying to find the logs, I have send a print on 3.2.1 and Seems like is working

I will let you know THANKS!!!!

Ghostkeeper, 2018-03-28:
Looks like it was fixed with Cura 3.2 then.

closed 2018-03-28 by Ghostkeeper