Ultimaker/Cura #3390: Support Z Distance is not applied to Support Top Distance

Horitsu, 2018-02-24:
If you change the Support Z Distance, then the value will now only applied to the Support Bottom Distance and NOT to the Support Top Distance. 

I observed this change of the support z/top distance behavior at changing from cura 2.7 to 3.1 and i'm using actual cura 3.2.1 (which has still this problem)

(in 2.7 was all ok, but in 3.1 and following it was broken)

ianpaschal, 2018-02-26:
Could you please provide more detailed steps and screen shots of what you're experiencing? I have just checked this on 3.2.1 and cannot recreate the issue. Support top and bottom distance are working perfectly for me.

closed 2018-03-09 by ianpaschal

Ghostkeeper, 2018-03-16:
The information provided is sufficient, really.

But I can't reproduce the issue. It seems to work all right as far as I can see. Here I've set the Support Z Distance to 1mm. Initially the Support Top Distance would then become 1mm and the Support Bottom Distance would become 0.5mm and that worked. Then I overwrote the Support Bottom Distance to 4mm and that works as well:
[image]