Ultimaker/Cura #3391: Cura 3.2.1 Spiral

Thisismydigitalself, 2018-02-24:
Cura 3.2.1.
Trying to "vase" this https://www.thingiverse.com/thing:494249 but even thou Spiralize settings (two) checked - Cura wont go into vase mode.

diegopradogesto, 2018-02-26:
Could you fill the template and send us a project file (File > Save Project...)?. With the project file we can check what happen because we have your machine, the model and the settings you have activated. Otherwise it is difficult to reproduce.

Thanks.

Thisismydigitalself, 2018-02-26:
Thank you. see above.

diegopradogesto, 2018-02-26:
The layer view just show layer by layer, it is difficult to show in a different way (continuous mode). The engine also works slicing the models layer by layer. I think the spiralize setting is working fine, at least is the same as I saw some many other times.
The only weird thing I saw in the gcode is that there is a ZHop at layer change. This is happening because the "Z Hop when retracting" is enabled at the same time as the "Retract at layer change" is also enabled. If you turn those off, then the print should go OK.
But the layer view will still show the same.
Let me know if this works for you.

Thisismydigitalself, 2018-02-26:
Thank you. i don't understand - I can't preview "Vase mode? my print came out not in vase mode.
Can you assure me that what you said which is "If you turn those off, then the print should go OK" i will be OK blindly? my point is: When turning Spiral ON, it should Spiral without dependencies or based on other settings to be turned OFF.

diegopradogesto, 2018-02-26:
Sorry, I was wrong, the two settings that I've indicated before don't affect the spiralize contour. They just affect at the first layers since you have the bottom layers. Once the bottom layers finish, then the rest of the model is printed in a "Vase model" as you call it. You can see that in the GCode the Z is increasing continuously:

`G1 X-19.755 Y58.123 Z0.601 E3310.39807
G1 X-22.278 Y58.033 Z0.602 E3310.44005
G1 X-24.842 Y57.991 E3310.4827
G1 X-27.003 Y57.97 Z0.603 E3310.51864
G1 X-29.586 Y57.932 E3310.5616
G1 X-32.565 Y57.841 Z0.604 E3310.61116
G1 X-34.665 Y57.709 E3310.64616
G1 X-35.059 Y57.684 E3310.65272
G1 X-39.043 Y57.141 Z0.605 E3310.71959
G1 X-39.812 Y57.029 E3310.73251
G1 X-41.999 Y56.473 Z0.606 E3310.77004
G1 X-44.029 Y55.723 E3310.80603
G1 X-45.879 Y54.754 Z0.607 E3310.84076`

Anyway, in the layer view we don't have a "vase mode" viewer, but the print should be as you expect.

Thisismydigitalself, 2018-02-26:
Thank you!
Can you please take a look at the genereted gcode i used to print a small sample and tell me if you see Spiral? this file was printed in vase mode but during printing it jumped each layer to the next one.
[L_vase.txt](https://github.com/Ultimaker/Cura/files/1758684/L_vase.txt)

diegopradogesto, 2018-02-26:
It seems to be working well. You can see this code indicating that Z increases continuously:

`G1 X-1.004 Y6.694 **Z45.593** E386.46901`
`G1 X-1.213 Y6.497 **Z45.595** E386.48334`
`G1 X-1.817 Y5.858 **Z45.6** E386.52721`
`;LAYER:151`
`G1 F1500 E381.52721`
`;TYPE:WALL-OUTER`
`G1 F1500 E386.52721`
`G1 F640.4 X-2.009 Y5.686 **Z45.601** E386.54007`
`G1 X-2.126 Y5.599 **Z45.602** E386.54735`
`G1 X-2.241 Y5.519 **Z45.603** E386.55434`

Probably you will still see some jumps in the first layers because there are bottom layers and those are printed in the common way.

To verify next time, you can open the gcode by yourself and check around the middle of the file to see if the Z slowly increases :-)

Good luck!

Thisismydigitalself, 2018-02-26:
Case closed? How come?

diegopradogesto, 2018-02-26:
I close it because I thought it was just a misunderstanding of those settings. However this is different. I didn't notice that overlap, but now I can see it. This looks like a bug.

Thanks!

Thisismydigitalself, 2018-02-26:
Great printers print alike.
Thanks.

Thisismydigitalself, 2018-02-26:
Just finished a new Spiral print, this time I manually disabled all retraction on layer change and Zhop and whatever i though might be related and print came out very nice. this means Vase mode works but feature must ignore retracts by default (like with your case https://github.com/Ultimaker/CuraEngine/pull/694)

Thx.

BagelOrb, 2018-03-06:
Closing this issue because the PR is already being handled.

closed 2018-03-06 by BagelOrb