Ultimaker/Cura #3392: [OT] Start analyzing sources

GMagician, 2018-02-24:
I would like to analize source code to be able and see if I can do something to help community.
I use C# and C so python may be something that needs to be learned.

Usually I develop using VS. I've already installed python addin but I need to prepare the project to let it compile but I think need to create project tree...

Is there somewhere something where I can start reading from?

Thanks

ianpaschal, 2018-02-26:
That's great that you'd like to help!

I recommend taking a look at the wiki, particularly at trying to build from source. There's continually improving documentation there on how to run Cura from source.

I would recommend thinking about something you want to do, like a particular plugin idea, or a certain bug that has been bothering you, and then try to to tackle that one thing.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-14:
This is my common reference for what g-code commands do: http://reprap.org/wiki/G-code

That coupled with applications like Cura and RepetierHost that can visualise g-code is how I got to learn to read g-code.

GMagician, 2018-03-14:
gcode is not an issue, it's my job and I work every day with it.
Phyton is unknown to me and use it in Visual studio as well since there is no info how to create a project with it.
At the moment I have the project but I need to add external classes (UM is one).
Wiki for windows run source is just....empty

Ghostkeeper, 2018-04-13:
Ah, sorry. We misunderstood you there.

I don't know about how Visual Studio works with Python, but if it has syntax highlighting and autocomplete you're probably fine. There's plenty of tutorials for learning Python and I can't really help you there except refer you to a couple of them. There's generic Python tutorials such as this one that's pretty well known: https://www.learnpython.org/ Myself I've learned the programming language via [Udacity](https://udacity.com/) but that takes more time.

It's probably easiest to start with writing your own post-processing scripts, that process the g-code after it's been generated. You can get some cool effects there, but you could start by modifying a simple one such as the [filament change](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/FilamentChange.py) script.

A bit more advanced, you could also try your hand at creating a plug-in for Cura. We've created a [few](https://github.com/Ultimaker/UraniumExampleExtensionPlugin) [examples](https://github.com/Ultimaker/UraniumExampleToolPlugin) to get you started. They are well documented.