Ultimaker/Cura #3393: Cura ignoring speed settings and going too fast

riddellriddell, 2018-02-25:
**Application Version**
Cura version 3.2 and also present in Cura version 3.1 

**Platform**
error occurred on a windows 7 PC

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
 Un sure how to repo but I think the error my have occurred when I opened up an old 3MF file in cura 3.1 

**Actual Results**
cura seems to ignore speed settings and bases movement off acceleration settings only, any strait movement happens almost instantly while any curved movement happens quite slowly. I have tried un installing and re installing, deleting user settings and reverting print settings to stock settings but nothing seems to fix it. speed and acceleration setting seem to be completely ignored.
I have attached an example of an object sliced with this error.

[Flawed print speed profiel.zip](https://github.com/Ultimaker/Cura/files/1755491/Flawed.print.speed.profiel.zip)

**Expected results**
consistent speeds around 45 mm per second 

**Additional Information**

closed 2018-02-25 by riddellriddell

tablatronix, 2019-02-10:
What was the issue, my printer just starting printing a job at super fast speed, no idea what happened.

Ghostkeeper, 2019-02-12:
Check if your initial layer speed is not set to something extraordinary, maybe. Or see if the problem is actually in the g-code or if it's just the printer itself doing something weird with the way it calculates step count per millimetre.