Ultimaker/Cura #3394: Print to USB Button Submenu

julianmallett, 2018-02-25:
**Application Version**
3.2.1

**Platform**
Mac osX 10.13.3 Sierra

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Plug USB cable in to Anycubic Kossel Delta, goto to select print to USB.

**Actual Results**
 The selection scroll button is off to the very top right hand of the screen and not immediately obvious.

**Expected results**
Would expect the choice between print to file and print to USB to be visible near the print button.

**Additional Information**
[image]

fieldOfView, 2018-02-25:
This should go away after resizing the window. It is something that has happened sometimes, but it happens in the UI framework we use which makes it hard for us to fix, other than shipping with an upgraded version of the framework.

julianmallett, 2018-02-25:
Many thanks, that worked for me.
have a great day

Julian

closed 2018-02-26 by ianpaschal