Ultimaker/Cura #3395: Non-Uniform Scaling Misbehaviour

ChrisRiddell, 2018-02-25:
**Application Version**
3.2.1

**Platform**
Windows 10

**Display Driver**
AMD 18.2.1

**Steps to Reproduce / Actual Results**
Add a cube turn off Uniform Scaling and scale Z, X or Y then try and scale it back once the mouse goes passed the gray marker it start scaling the wrong way.

[image]

**Expected results**
It keep scaling the correct way.

Ghostkeeper, 2019-06-10:
Fixed in 4.0.

Ghostkeeper, 2019-06-10:
Ah, nevermind. I misunderstood. The scale goes at the proper speed now, but still becomes larger when going beyond the origin since it just depends on the distance to the origin.

GregValiant, 2024-11-14:
This does not seem to be a problem in current versions of Cura (5.8.0 and up).  I'll go ahead and close this.

closed 2024-11-14 by GregValiant