Ultimaker/Cura #3396: Model with mesh type modify settings still have walls and top/bottoms

ChrisRiddell, 2018-02-25:
**Application Version**
3.2.1

**Platform**
Windows 10

**Display Driver**
AMD 18.2.1

**Steps to Reproduce / Actual Results**
Open the Project file or load a model, add cube use mesh type modify settings for infill of other models, placing the mesh type cube into the model and slicing you will see it still has walls and top/bottom layers. 

[image]

[UM2_cube.curaproject.zip](https://github.com/Ultimaker/Cura/files/1755935/UM2_cube.curaproject.zip)

**Expected results**
The Mesh types to only change/effect the settings that are changed for that type not do things like have its own walls and top/bottom layers etc.

smartavionics, 2018-02-27:
I think you still have to explicitly set the number of walls and skins to zero for the infill mesh.

ChrisRiddell, 2018-02-27:
If you do that and the mesh type model overlaps the walls of the main model it turns off the walls on the main model though.

smartavionics, 2018-02-28:
Ah, yes, it would do that so you need to ensure the infill mesh isn't oversize.

ChrisRiddell, 2018-02-28:
Been playing around and this happens with all mesh types not just infill i'd expect it to only effect/change what ever setting was changed to that mesh type model.

Anther eg is fuzzy skin if I use a mesh type to turn off fuzzy skin i'd expect where I overlap the model to only turn off the fuzzy skin and not add walls etc.

BagelOrb, 2018-03-06:
This is expected behavior of the current implementation. The setting Top/Bottom Thickness of that mesh is still a positive number.

It would be nice if selecting the infill mesh type automatically loads the following per object settings:
- Wall Line Count = 1
- Top/Bottom Thickness = 0

Calling on the omnipresent benevolence of the almighty @fieldOfView

printingotb, 2019-06-24:
Has this issue received any more attention? I would love to be able to change settings for an overlap without it dividing the model into two, with walls printed between them.
In this example, I wanted to print the outer wall slow on a part of a sword model to achieve a shiny surface on the edge only. However, doing this adds a lot of unnecessary material and will create a seam where the parts intersect.
Just one example.
[image]

BagelOrb, 2019-06-25:
I don't think this issue ever got more attention. Perhaps I will look into it somewhere in the following years.

Ghostkeeper, 2019-06-26:
It's touching a limitation of our data model where we store those settings per object, not per line. Thus for walls to have a different speed, they need to be walls of a different object. If they are a different object, they will get walls around the whole thing in a closed loop. It didn't get attention because it'll be a lot of work to fix.

printingotb, 2019-06-26:
Thanks for the explanation.

GlOwl, 2019-10-24:
I have almost the same issue as printingotb just that I want to change the support settings and not printing speed. i am not familiar with the code, but is the data model preventing to implement it for supports and other not object altering settings? Or are support settings included in the object too?

And I'm using cura 4.3 on linux (so its not a windows specific issue ;) )

BagelOrb, 2019-10-24:
@GlOwl You might need to look into support blockers. I don't think this is related. Please ask your question in a new thread.