Ultimaker/Cura #3397: 3.2.1 USB printer detection

macaron16, 2018-02-25:
**Application Version**
3.2.1

**Platform**
OsX High Sierra
NVIDIA GeForce GT 650M 512 Mo

**Steps to Reproduce**
Printer is On, Usb cable not connected
Connect the Usb cable.
Printer isn't recognized.

Close Cura and restart it.
Printer is recognized.

slundell, 2018-02-26:
I experienced the same thing with my Cetus3D/Smoothie printer. It is due to that the printer needs some time to init, but Cura tries to correct it straight away. This is fixed in my PR #3346 . If you are able to test it out, that would be great as I can only test on the printer I have available.

macaron16, 2018-02-26:
Sorry, I'm not able to do it. I don't know where to find the file to edit in OSx.

ianpaschal, 2018-02-27:
@macaron16 You'd need to run from source.

Unfortunately, at Ultimaker we don't have non UM machines to test on. We'll have to see what happens with your PR @slundell .

ianpaschal, 2018-02-27:
Oops did not mean to close.

Ghostkeeper, 2018-03-14:
> Sorry, I'm not able to do it. I don't know where to find the file to edit in OSx.

Slundell's PR has been merged, so once the 3.3 beta is released, you can test it using that beta.

ianpaschal, 2018-03-14:
> Slundell's PR has now been merged

In that case I'm closing this. Otherwise we end up with lots of open issues that were actually fixed but no one remembered/had time to close them.

emaayan, 2018-10-09:
this still happening, and is VERY annoying, in the event i need to do abort print or reset the printer, it's one of the things i'm still on s3d, at least have a connect button

Ghostkeeper, 2019-03-07:
We need to know what is different with your printer that makes it not connect with yours but does connect with other printers. We don't have any Cetus3D/Smoothie printer here, so we can't test anything.

diegopradogesto, 2019-07-02:
Is this still happening in the latest versions of Cura (3.6, 4.0 or 4.1)?. @emaayan, @slundell, @macaron16 could you please confirm?

Closing this in the meanwhile, since it was not updated from a long time now.

closed 2019-07-02 by diegopradogesto