Ultimaker/Cura #3399: 3.2.1 Usb Print

macaron16, 2018-02-25:
**Application Version**
3.2.1

**Platform**
OsX macOs High Sierra

**Qt**
**PyQt**

**Display Driver**
NVIDIA GeForce GT 650M 512 Mo

**Steps to Reproduce**
Print on USB

**Actual Results**
During printing by USB, there are false commands send to the printer.
X/Y movement, and certainly Z and E are not required !
For sample, X movement goes to home, or Y movement goes to an oversize value during printing the skirt or the first layer.
--> result is awful --> Stop immediately.

It's seems that "Start gcode" is ok every time.

**Expected results**
printing in the same way than "save the gcode to file" on the SD Card.

**Additional Information**
Printing on an ANET A8, with CH341SER_MAC driver (CH34x_Install_V1.4.pkg)

ChrisTerBeke, 2018-02-28:
The G-code that's send over USB is the same as that's stored in a G-code file when exporting, so if there's a difference between those two printing methods but slicing both with the same settings in Cura, I suspect it's a firmware/printer issue, not a Cura issue.

macaron16, 2018-02-28:
It could but it's also possible that cura send to quickly some "codes" that generated missing commands.

ChrisTerBeke, 2018-02-28:
Cura sends the next command when the printer sends an "OK" back. The printer decides how much to buffer. I have an Anet A6 myself and the stock firmware is just horrible (re-flashed it with custom marlin to make it work). Given that most other USB printers work fine with Cura (even though it's not a recommended way of printing), I still think it's not a Cura issue.

macaron16, 2018-03-01:
ok

macaron16, 2018-03-01:
So, I upgrade the firmware to marlin 1.1.8.
No issue with this firmware. That means it isn't a cura issue.

closed 2018-03-06 by BagelOrb